AlexandreRouma/SDRPlusPlus #1148: SDR++ server RaspberryPi version only runs as user id 1000

marrem, 2023-07-22:
**Hardware**
RaspberryPi 3B+
SDRPlay RSPdx

**Software**
-  OS: Raspbian GNU/Linux 11, Linux mpw 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
- SDR++ v1.1.0

**Bug Description**
When running as user id 1000 ('pi' on older versions of Raspbian, user supplied name when installing newer versions of Raspbian (in my case 'marc')): sdr++ server runs fine. I can connect to it from a remote laptop running SDR++.

I want to install sdr++ as a systemd service. I created a user sdrpp. When I run `sdrpp --server` as sdrpp, I can connect to it, but when I start the SDR, the server complains:

```
[22/07/2023 17:25:25.000] [INFO] AudioSourceModule 'Audio Source': Menu Select!
[22/07/2023 17:25:25.000] [INFO] Ready, listening on 0.0.0.0:5259
[22/07/2023 17:25:30.000] [INFO] Connection from TODO:TODO
[22/07/2023 17:25:31.000] [ERROR] Error opening audio device: std::exception
[22/07/2023 17:25:31.000] [INFO] AudioSourceModule 'Audio Source': Start!
```

Just to try I ran `sdrpp --server` as root. Even then, it complains about not being able to open the audio device.

How does user 1000, trigger sdr++ server to use SDRplay instead of AudioSource?

This is stdout with user 1000 ('marc'):

```
[22/07/2023 17:34:59.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Menu Select!
[22/07/2023 17:34:59.000] [INFO] Ready, listening on 0.0.0.0:5259
[22/07/2023 17:35:04.000] [INFO] Connection from TODO:TODO
[22/07/2023 17:35:08.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Tune: 120575000.000000!
[22/07/2023 17:35:09.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Start!
[22/07/2023 17:35:09.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Tune: 120575000.000000!
```

It seems that somehow with user 1000, it correctly tries to use SDRplay, but with other users it tries to open AudioSourceModule.

**Steps To Reproduce**
1. Run `sdrpp --server` as a user, not being id 1000, the default user on a Rpi.
2. Connect to it from SDR++
3. Observe that no signal appears in SDR++
4. Observe the message: [ERROR] Error opening audio device: std::exception on the server's stdout.

AlexandreRouma, 2023-07-22:
Not a bug.
The server and client both load their configuration from the user directory `~/.config/sdrpp`.
Since you cropped the logs (contrary to what's asked for in the template) I can't really tell you more.

marrem, 2023-07-22:
Yes. I was about to close this issue. I saw that the difference was in .config/sdrpp/config.json "source". I can't remember having set this on installing sdr++ on the raspi. But I saw that what set from SDR++ (client) is written to config.json. So I forgot that I chose SDRPlay there....

closed 2023-07-22 by marrem