AlexandreRouma/SDRPlusPlus #1149: rtaudio-6.0.0

srs4511351, 2023-07-25:
**Feature description**
rtaudio-6.0.0 has been released, but SDR++ does not work with it.

When compiling, cmake found rtaudio, version 6.0.0

make has this error:
```
/home/dietpi/SDRPlusPlus/source_modules/audio_source/src/main.cpp: In member function ‘void AudioSourceModule::refresh()’:
/home/dietpi/SDRPlusPlus/source_modules/audio_source/src/main.cpp:93:26: error: ‘struct RtAudio::DeviceInfo’ has no member named ‘probed’
   93 |                 if (info.probed && info.inputChannels < 2) { continue; }
      |                          ^~~~~~
```
Will you consider making SDR++ work with rtaudio-5.2.0 and rtaudio-6.0.0?

----Steve

AlexandreRouma, 2023-07-27:
I can't support 6.0 without dropping support for earlier versions since the author of RtAudio didn't add any compile-time way to check. Since almost all distros use rtaudio 5.2 and earlier, I won't support 6.0

srs4511351, 2023-07-27:
cmake did list the version that it found. It said "Found rtaudio, version 6.0.0'

Some of the other apps like CubicSDR works with either version.

Maybe your best action is to add a warning in readme.md

----Steve

srs4511351, 2023-09-22:
Thanks for offering to work on it.
In my opinion, if you can achieve the performance of RtAudio without it, you don't need it.
I will have it installed for the other SDR applications that need it.

phoniclynx, 2024-01-10:
> Dear AlexandreRouma, I am writing to you because I am using Manjaro (derived from Arch) and I am having problems with RtAudio. Since Manjaro was rolling-release, it was updated to RtAudio 6 and SDR++ stopped working. Now I downgraded to RtAudio 5.2.0, compiled and SDR++ works like a charm.
> 
> I think this may be a temporary solution for those of us who have a rolling-release system that updates too much.
> 
> Best regards, Matías Saibene.

What did you do to install RtAudio 5.2.0? What was the command line to do it? (I'm in Manjaro too)

AndreaVigarani, 2024-01-13:
> What did you do to install RtAudio 5.2.0? What was the command line to do it? (I'm in Manjaro too). I can only see an rtaudio 6 and no version 5 listed in the AUR

If you do not need rtaudio newer version for something else you can use
"manjaro-downgrade rtaudio"

Mr-Precise, 2024-01-15:
I added the rtaudio 5.2 ([rtaudio5-exp](https://aur.archlinux.org/packages/rtaudio5-exp)) package to the AUR, sdrpp should now build.
Use with caution, it maybe break "backward compatibility" with other software...
Any suggestions welcome.

srs4511351, 2024-01-15:
If other programs were compiled with rtaudio 6, they may have to be reinstalled and compiled with rtaudio 5.2 installed. Some may work as is.

closed 2024-01-22 by AlexandreRouma

srs4511351, 2024-01-24:
I suggest that in you state in the instructions that SDR++ will work with the version of RtAudio that it was built with, 5.x or 6.x.
You can't just upgrade RtAudio and expect it to work with SDR++

AlexandreRouma, 2024-01-24:
You realize that's how all compiled software works on unix-like systems right?

srs4511351, 2024-01-24:
No, CubicSDR worked with RtAudio 6.0.1, having been compiled with RtAudio 5.2

In addition, not everyone knows that, it only takes a few words.