AlexandreRouma/SDRPlusPlus #1154: arm64 support

RichNeese, 2023-07-30:
Can you please work on a arm64 release ?
alotof us on arm are playing with sdr

AlexandreRouma, 2023-07-30:
arm64 what? macOS? Windows? Linux?
I only one a single raspberry pi and it does the armhf builds. Github's CI still doesn't support M1/M2 so I can't compile for that either.

RichNeese, 2023-07-31:
Arm64 linux not armhf…. But true arm64 linux support

Mr-Precise, 2023-08-02:
I tried to build the Linux arm64 version of sdr++ (by running the original bullseye arm64 rpi image in virtualization).
I didn't check the functionality, I don't have Raspberry Pi hardware...
[Mr-Precise/SDRPlusPlus/nightly-arm64](https://github.com/Mr-Precise/SDRPlusPlus/releases/tag/nightly-arm64)
I'm planning to buy a Raspberry Pi 4, and I'm also interested in whether it will work

AlexandreRouma, 2023-08-21:
> Arm64 linux not armhf…. But true arm64 linux support

Read what I said again. I'm telling you the only existing arm build is armhf, I know you want arm64 which is different.

RichNeese, 2023-08-22:
NO I have built on arm64 . I built from src and it builds and runs fine.

IComplainInComments, 2023-09-03:
> NO I have built on arm64 . I built from src and it builds and runs fine.
> […](#)

The point he is making is Github's CI doesnt support building for M1/M2, and the build system he uses for ARM is a raspberry pi on 32bit. As such he can't make packages since he does not have access to a system that builds for 64bit ARM (ARM64).

qrp73, 2023-09-04:
it is already supported on aarch64, currently I'm using it on aarch64 linux and it works very well

w1ebr, 2023-09-19:
Is an anyone running SDR++ on a Mac / M1 or M2 ? I am trying to plot this to iPadOS for the iPad Air and Pro models that use the Mx chips. Building the Volk library is where I need help at the moment. If someone has built this for a Mac running aarch64, would you please send me the Volk static libraries (e r I s I  at  c o m c a s t . n e t)

H6LS1S, 2023-09-24:
> Is an anyone running SDR++ on a Mac / M1 or M2 ? I am trying to port this to iPadOS for the iPad Air and Pro models that use the Mx chips. Building the Volk library is where I need help at the moment. If someone has built this for a Mac running aarch64, would you please send me the Volk static libraries (e r I s I at c o m c a s t . n e t)

I use the sdr++ client for intel on M1, it works quite well, but there are some instabilities in the client and sound server

RichNeese, 2023-09-24:
volk is in brew. brew search volk

its there

johntdavis84, 2023-09-24:
Does installing the Volk package have any advantage when running SDR++ in client mode with a server on a remote Linux box, or is it just helpful for using the server mode?
I’m actually a bit unclear on what Volk does.

w1ebr, 2023-09-25:
@H6LS1S , I want to run this on an iPad so I would prefer not to run in an x86 emulator.

@RichNeese , I have installed Volk via brew. It says it is available for both Intel and aarch64 architectures. When I specify the architecture using the brew "fetch" command, I can't find an aarch64 version, just an x86/64 version. I will try again, but if you have any suggestions,  please  let me know.

@johntdavis84 , Volk is two things; a specification of a set of DSP - type vector instructions and it is an implementation of the spec on different processors. This allows Volk users to focus on the mathematical functions and not on the hardware specific implementation of those functions.

If the SDR++ server is streaming I/Q data, then Volk is important to the client function.

IComplainInComments, 2023-10-09:
> Does installing the Volk package have any advantage when running SDR++ in client mode with a server on a remote Linux box, or is it just helpful for using the server mode? I’m actually a bit unclear on what Volk does.
> […](#)

Volk is a cross-platform Vector optimization library, so when we are using FFT (Fast Fourier Transform) Volk would help a lot. Basically the project hand-tunes mathematical operations to properly and efficiently use SIMD instructions.

Although personally, I say use the GPU for this using OpenCL, but I'm not sure what specific goals you have for the program as a whole when processing wave forms.

AlexandreRouma, 2023-10-09:
> I am trying to port this to iPadOS for the iPad Air and Pro

Short of writing an entire render backend for iOS, you won't get this running. In any case, apps don't have the raw access to USB ports that is needed for SDR hardware to work, and is why little to no SDR software is available on the platform. The few that you can run are remote clients only.

> Does installing the Volk package have any advantage when running SDR++ in client mode with a server on a remote Linux box, or is it just helpful for using the server mode?
I’m actually a bit unclear on what Volk does.

Volk does all the SIMD abstractions to make the basic operations in the DSP code such as addition, multiplication, dot product, etc much faster on every platform that supports vector instructions.

> Volk is a cross-platform Vector optimization library, so when we are using FFT (Fast Fourier Transform) Volk would help a lot. Basically the project hand-tunes mathematical operations to properly and efficiently use SIMD instructions.

Not only is volk not used to compute the FFT but it doesn't even have that functionality to begin with.

> Although personally, I say use the GPU for this using OpenCL, but I'm not sure what specific goals you have for the program as a whole when processing wave forms.

OpenCL has way too much overhead to do live signal processing or even render the FFT, which is one of the least CPU heavy tasks the software does anyway. Not to mention systems with only integrated GPUs...

IComplainInComments, 2023-10-09:
> Short of writing an entire render backend for iOS, you won't get this running. In any case, apps don't have the raw access to USB ports that is needed for SDR hardware to work, and is why little to [...]

Actually for iPadOS this is possible, as you are able to use drivers that are written in [DriverKit](https://developer.apple.com/documentation/driverkit/creating_drivers_for_ipados).

> OpenCL has way too much overhead to do live signal processing or even render the FFT, which is one of the least CPU heavy tasks the software does anyway. Not to mention systems with only integrated [...]

That makes sense actually, although integrated GPUs would do just fine in all honestly, any in the past 6 years easily hit 1TFLOP. Just to clarify though, I do agree the CPU is just fine, especially after you defined the scope.

AlexandreRouma, 2023-10-09:
> Actually for iPadOS this is possible, as you are able to use drivers that are written in [DriverKit](https://developer.apple.com/documentation/driverkit/creating_drivers_for_ipados).

No SDR device has this sort of drivers. Almost all SDR use libusb for their driver which is why they need raw USB access.

IComplainInComments, 2023-10-09:
> No SDR device has this sort of drivers. Almost all SDR use libusb for their driver which is why they need raw USB access.

Now im sure some work would need done to adjust this for use with DriverKit, but it has already been done for a LONG time... [libusb for iOS](https://github.com/Brandon-T/iOUSB).

IComplainInComments, 2023-10-09:
> Now im sure some work would need done to adjust this for use with DriverKit, but it has already been done for a LONG time... [libusb for iOS](https://github.com/Brandon-T/iOUSB).

Look in the README under dependencies, there is an explanation.

AlexandreRouma, 2023-10-09:
Feel free to contact every device manufacturer about this if you want to spend months getting every driver working with it.
Note that even on android , where libusb can be used natively, it took me several weeks to get only a subset of devices working properly. Have fun!

w1ebr, 2023-10-10:
It is a huge challenge and I have doubts about whether I can port it to the iPad. My goal is to be able to use the USB port on my iPad with a USB C connector. As you said, even with OpenGLES (which is depreciated but still available) a backend has to be created and I have spent the last few weeks just trying to get things to compile and link properly. The iPad hardware is nice but porting to iOS is a challenge.GeneSent from my iPad

w1ebr, 2023-10-30:
An update on my status fwiw: I am looking into using SDL and Metal as the renderer for imgui. I have compiled versions of volk, sdl2 and fftw for iPadOS. SDR++ uses a mix of imgui and OpenGL calls which is why I thought OpenGLES would be the way to go, but it is depricated in iOS. I have decided that I need to take a few steps backward if I hope to be able to move forward, so I will first work on a version that uses SDL and Metal on a MBP. Once that is working, porting to iOS should be much more straightforward for the core function. Getting the USB interface working fully is something that I have no way to check out. I do have an RTL dongle and that’s where I hope to start.
Gene

alphaguy, 2023-12-16:
I am trying to run SDR++ on a new PI5 running a 64 bit OS after bullseye, which is bookworm.  Any help would be appreciated.  I would really rather not build from source but I will if that is the only way to get there.

AlexandreRouma, 2023-12-16:
it's the only way

jerrynsr, 2024-02-06:
Look in my fork of the Master repo.  https://github.com/jerrynsr/SDRPlusPlus/tree/patch-1   I built a deb package for Bookworm.
To build all the plugins I need two missing files.    pll/dsp.h and pll/demodulator.h      I have been looking at older repos to merge into my checkout code to test compile but I have not found it.  It's almost as if this version is a rewrite and not all the code is merged in.

darksidelemm, 2024-12-03:
Was this not happening because there were no Github runners for arm64? These are now definitely available, I'm using them in a few projects...

waltje, 2024-12-15:
@darksidelemm yes, that is why I was asking again in #1546 ;-)

geerlingguy, 2025-02-13:
Putting my hat in the ring for a nightly for arm64 Linux (main target for me would be Pi 5 / Debian 12 Bookworm, but also for Ampere systems, or other Arm boards like Radxa/RK3588 SBCs, or the new Orion O6 motherboard with a Cix chip).

Arm64 runners on GitHub Actions for CI are also available now (though I haven't spent much time messing with them): https://github.blog/changelog/2024-09-03-github-actions-arm64-linux-and-windows-runners-are-now-generally-available/ (not sure if that was a hindrance to getting the build working).

AlexandreRouma, 2025-02-13:
Did hear about the Github-hosted aarch64 runners back when they were announced but wasn't able to use them for some reason (guess they weren't public yet?).

At the moment every 32bit build is done on a runner hosted on my only modern raspberry pi (pi 4 2GB), which isn't the fastest thing ever...

I'll see If I can get the github hosted aarch64 runner working. It would indeed allow for a 64bit ARM debs to be available and remove the need for my poor raspberry pi to do all the work :)

AlexandreRouma, 2025-02-13:
@geerlingguy Could you try the artifact name `sdrpp_debian_bookworm_aarch64` from [this new CI run](https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/13318493805)? It should install like any deb, just make sure to install the drivers for the SDR you want to use.

You'll notice that when you open the zip, the deb is named `sdrpp_debian_amd64.deb`, don't worry about the "AMD64" part, it's just because I copy pasted the docker script for it from the regular amd64 CI environment, it **should** be aarch64.

Note that the cariboulite won't work as SoapySDR support has been deprecated from SDR++, so use something like a RTL-SDR or a HackRF. I do have a cariboulite here and planned to add support but it's a pre-production prototype which doesn't seem to work with the published driver :/

waltje, 2025-02-13:
Or, do what I do- install cross-compile envs for such platforms.  

--Fred

 
  

At the moment every 32bit ARM build is done on a runner hosted on my only modern raspberry pi (pi 4 2GB), which isn't the fastest thing ever...

At the moment every 32bit ARM build is done on a runner hosted on my only modern raspberry pi (pi 4 2GB), which isn't the fastest thing ever...

waltje, 2025-02-13:
Yeah, I parametrized that script for that reason.  It defaults to amd64.

AlexandreRouma, 2025-02-13:
> Or, do what I do- install cross-compile envs for such platforms. 

Good luck getting cross-compiling working properly with a project like this using pkg-config for finding dependencies.

> Yeah, I parametrized that script for that reason.  It defaults to amd64.

I don't know what script you're talking about, everything involved in the CI was written by me.

waltje, 2025-02-13:
> Good luck getting cross-compiling working properly with a project like this using pkg-config for finding dependencies.I ccompiled sdrpp on a linux box just fine, for my arm64-based boards.  There was one module that gave trouble,cannot remember which one.> don't know what script you're talking about, everything involved in the CI was written by me.make_debian_build.sh.

AlexandreRouma, 2025-02-14:
aarch64 debs have been added to the nightly release.
I'll wait for more beta testers to come through with feedback and if everything is fine I'll close this issue.

geerlingguy, 2025-02-14:
I will try to test in the next day or two; been busy with some personal matters this week otherwise I'd get on it right now, haha!

I am eager to check out SDR++, and have a remote Pi running with Pi Connect with an RTL-SDR v4 plugged in.

waltje, 2025-02-14:
I built it for aarch64 just fine, so there is no reason why the nightlies would not work.  I run themon rpi4, 5 and two non-rpi aarch64 boards.  

Will grab the nightly tomorrow and try it.

geerlingguy, 2025-02-17:
I tested out the nightly from 4 days ago:

```
cd Downloads
wget https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/nightly/sdrpp_debian_bookworm_aarch64.deb
sudo apt install -y ./sdrpp_debian_bookworm_aarch64.deb
```

And it's working a treat!

[image]

AlexandreRouma, 2025-02-17:
Alright perfect! The arm64 build are now available from the regular nightly release.
Since it seems they work properly, I'll close this issue as resolved.

Also side note, I know you probably just used defaults just to check, but just in case, for best results with FM broadcast demod in the US, use the following settings:
- Bandwidth: 200k to 250k
- Deemphasis: 75us
- IF noise reduction: **OFF**
- Stereo: on
Then you can also turn on RDS decoding to see the

closed 2025-02-17 by AlexandreRouma

waltje, 2025-02-17:
Seconded on my rpi4 and rpi5
Have not tested on the non-rpi boards, but should work, as those, too, are using a bookwork distro.
--fred

 
  

Screenshot.2025-02-17.at.11.09.35.AM.png (view on web)

Screenshot.2025-02-17.at.11.09.35.AM.png (view on web)