AlexandreRouma/SDRPlusPlus #1157: RIGCTL client in latest nightly build version

ON4BCB, 2023-08-02:
I did some testing with the latest nightly build for windows.

I setup a raspberry Pi with Rigctl working with a FT817 radio

Rig CTL is working fine

When setting up the rigctl client in sdr++ i noticed that from time to time it only updates the radio's frequency when changing frequency in sdr++, (if i understood well it is only working in one direction which is fine for me)

To be shure i installed a local TCP server to test the connection localy on 127.0.0.1, here i also noticed the same problem
there is a frequency update if you make big changes in frequency or even frequency band but the frequency is not right (i changed IF to 0 in sdr++ so no offset)

Would it be possible to have a look to rigctl client maybe there might be a bug

Looking forward to test it whenever a new version is available

Walter

ly2ss, 2023-08-05:
I use AirspyHF+ as panadapter receiver in SDR++ (master)  and FT-991a as my rigctld connected rig. They both connected to the same antenna, **not via IF tap**.  I noticed that when I click Start button (connect to rigctld), my AirspyHF+ receiving frequency is instantly shifted to 0Hz (the same as IF frequency as I entered in IF window. , while waterfall's frequency stays the same as before connection. The frequency is pushed to rig correctly also:

[image]

Seems in this hardware configuration case with 0Hz  IF, the SDR++ should not shift Airspy's frequency.

73! Žilvinas LY2SS

AlexandreRouma, 2023-08-21:
There is no bug. The only configuration the rigctl client currently supports is receiving at a fixed IF and sending tuning commands to the transceiver instead.

closed 2023-08-21 by AlexandreRouma