AlexandreRouma/SDRPlusPlus #1162: USB soundcard not listed when more than one is connected.

magellan-13016, 2023-08-05:
All is in the title. I have two USB sound cards connected to my computer. On is integrated to the digital interface to control my radio with my computer and the other one is used for SDR++ audio output.

In the devices list of my computer, both are presents with unfortunately the same name : USB Audio device.
Because they have the same name, SDR++ choose the first one which is not the good for me.
[image]
[image]

SDR++ should show both so I can select the good one.

Thanks for reading. 

David.

AlexandreRouma, 2023-08-25:
I think it's because they have the same name. Even if I were to make sure they both appear there's go guarantee on the order so you should report the issue to whoever wrote the driver for your soundcard.

AlexandreRouma, 2024-01-27:
Investigating further, this is a bug inside RtAudio. This means nothing in SDR++ is wrong, so I'll close this issue.

The RtAudio based sink will be replaced in the coming weeks with a sink using a custom audio abstraction library which will get rid of this issue.