AlexandreRouma/SDRPlusPlus #1170: Add support for RTL-SDR Blog V4

pedroCX486, 2023-08-28:
**Copied and edited from here: https://github.com/f4exb/sdrangel/issues/1783**

Recently RTL-SDR Blog V4 was released (https://www.rtl-sdr.com/rtl-sdr-blog-v4-dongle-initial-release/) which is based on the R828D tuner chip. But the hardware is slightly different as it uses a different LO frequency for the R828D compared to what is hardcoded in the drivers, and the RF inputs need to be switched in a different way.

A fork of the Osmocom drivers were created here https://github.com/rtlsdrblog/rtl-sdr-blog-minimal, and a more heavily modified one here with other unrelated changes https://github.com/rtlsdrblog/rtl-sdr-blog. These repo's were modified to add support for the V4, and the -minimal fork has the minimal amount of changed to make it work. A patch to has been submitted to Osmocom to try and upstream the main changes for the V4, but the Osmocom rtl-sdr project might be dead as I see that patch requests have been unanswered for a few years.

If it's not upstreamed is there some other way to support the RTL-SDR Blog V4?

On Windows it's easy as I can just tell users to replace the .dll file, and on Linux users can just update the system RTL-SDR driver. But for MacOS and Android the driver gets shipped with the package, and it's a bit harder to update.

The forked drivers continue to support older RTL-SDRs and other brands/generic models too, nothing has been removed.

vornao, 2023-08-30:
It seems that Osmocom upstreamed changes last week! Please refer to [this link](https://gitea.osmocom.org/sdr/rtl-sdr).

pedroCX486, 2023-08-31:
> It seems that Osmocom upstreamed changes last week! Please refer to [this link](https://gitea.osmocom.org/sdr/rtl-sdr).

I assume now it's "just" a matter of updating the fetch/submodule from their repo?

AlexandreRouma, 2023-08-31:
I know, I work with RTL-SDRblog and already have the dongles.

On linux SDR++ does not bundle the driver so there is nothing to change on this end.

pedroCX486, 2023-08-31:
> I know, I work with RTL-SDRblog and already have the dongles.
> 
> On linux SDR++ does not bundle the driver so there is nothing to change on this end.

My issue was mostly in a general manner, but FWIW, I'm using SDR++ on a Mac.

pedroCX486, 2023-09-03:
> I know, I work with RTL-SDRblog and already have the dongles.
> 
> On linux SDR++ does not bundle the driver so there is nothing to change on this end.

@AlexandreRouma: My issue was mostly in a general manner, but FWIW, I'm using SDR++ on a Mac, which has the drivers embedded and requires a new build. I have both Intel and M-series Macs at my disposition, can test if needed.

pedroCX486, 2023-09-06:
https://github.com/AlexandreRouma/SDRPlusPlus/pull/1176 by me fixes this. Waiting review.

closed 2023-09-13 by AlexandreRouma