AlexandreRouma/SDRPlusPlus #1173: unity gain for spectrum fft

qrp73, 2023-09-04:
I found that the latest version build from github repository has inconsistent gain for different FFT size and window type. It should use correction which depends on FFT size and window coefficients in order to keep the same gain for any selected mode.

For example if we put the sine with amplitude 1.0f on the input, it should show FFT bin with peak magnitude 0 dB independently on selected FFT size or window type.

**Steps To Reproduce**
1. put sine with amplitude 1.0f on the input of SDR++ processing chain
2. try to change FFT size and window type setting and check if peak magnitude of FFT bin on display is not changed

Expected result: the peak magnitude of FFT bin remains unchanged when you change FFT size or window type

Actual result: the peak magnitude of FFT bin is changed when you change FFT size or window type

Could you please fix that?
Thanks

AlexandreRouma, 2023-09-04:
Not a bug.

The levels are lower when changing the FFT size because you're probably using an SDR that can't produce enough data to fill the FFT buffer at the rate you asked to draw the FFT. The buffer gets padded with zeros and windowed only in the filled region. This is to maximize time resolution at the cost of frequency resolution. Basic math...

And for the difference between windows, I just didn't bother normalising the windows. it's an additional compute step that few people would notice and even fewer would care about.

closed 2023-09-04 by AlexandreRouma

qrp73, 2023-09-04:
> The levels are lower when changing the FFT size because you're probably using an SDR that can't produce enough data to fill the FFT buffer at the rate you asked to draw the FFT.

I'm used SDR that produce samples at 960 kHz sample rate (1920 kHz bandwidth). I'm using HPSDR protocol with custom speed (I replace speed values in the hermes_source code for using it with my hardware). Regarding to not enough sample rate, I think this is not the reason, because it changes bin magnitude even when switching between FFT size 1024 and 2048. As you can see 960 kHz sample rate is well enough to fill FFT buffer at least 460 times per second. 

I will test it more deep later. Could you please help me to find where is the code which performs FFT and draws the spectrum?

> And for the difference between windows, I just didn't bother normalising the windows. it's an additional compute step that few people would notice and even fewer would care about.

Window normalization can be done only once when you calculate it's coefficients, so it won't affect performance. A lot of peoples using SDR for signal measurements (I think about 80-90% of these who using SDR and know what they are doing), and it affects them all...

Also there is a lack SNR meter in dB for measurements and dBFS level indicator which can help to control that input signal is within safe ADC working range and you can believe results...

AlexandreRouma, 2023-09-04:
> 960 kHz sample rate (1920 kHz bandwidth). 

That's not how math works.. 960KS/s IQ is 960KHz bandwidth, no way around it....

> FFT size 1024 and 2048

Prove it... Tested it *again* just now and it's not different
[image]

> I found that hermes_source uses some "link" stream and has "TODO" comments to fix it. Can it be related with this the issue?

The DSP and FFT literally have no idea what source is being used...

> Window normalization can be done only once when you calculate it's coefficients

I know, I'm not an idiot. I just don't bother with code that nobody cares about being there or not.

> Also there is a lack SNR

literally top right of the window....

> dBFS level indicator

CTRL+hover a VFO

qrp73, 2023-09-04:
>> CTRL+hover a VFO

Thanks, found it. But it shows "strength" in dBFS and it depends on the VFO position and bandwidth. This is not what I'm talking about. 

I'm talking about full signal dBFS level. Signal level dBFS shows the max peak sample amplitude relative to the min/max possible ADC value. Since you're using float samples, it should show the max peak sample in the stream relative to 1.0f value. This value doesn't depends on selected bandwidth, it shows how much gain is required in order to get clipping for input signal.

In other words this is indication how much voltage you can add to get ADC overload. Just test HDSDR software, you will find that dBFS level doesn't depends on VFO position and bandwidth. It depends on min and max sample values for entire input stream.

AlexandreRouma, 2023-09-04:
> this is a bad test case, you're needs to test it on stable and strong signal. Noise and air carriers with fading is a bad choice for testing it.

I looked at it switching back and forth and there was no difference. Why would I lie to you about this?
So again, if you say there is a problem, **prove it**. I'm not wasting time on circumstantial "I feel something is wrong" type of reports.

> relative to the min/max possible ADC value.

There is no point displaying this. Almost all SDRs do some kind of processing to the IQ that will change the raw level compared to the real ADC values meaning the value you'll measure will be basically useless. Having to run basically AM demod on the raw IQ wastes CPU just to show a meaningless ground for everyone else is silly. Moreover, if the ADC clips you will very much notice by the spectrum filling with image signals and the noise floor rising significantly.