AlexandreRouma/SDRPlusPlus #1178: RTL-SDR doesn't appear in the Source menu - MacOS

o-Dan-o, 2023-09-07:
**Hardware**
- CPU: Apple M1
- RAM: 16 GB
- GPU: Apple M1
- SDR: RTL-SDR V3

**Software**
- Operating System: Ventura 13.5.1 (22G90)
- SDR++: v1.1.0 03:44:09, Sep 6 2023 (nightly version)  sdrpp_macos_intel.zip 

**Bug Description**
USB RTL-SDR does not appear in the Source menu, although the USB dongle is recognized by the OS. Refresh does not work, nor does reinstall and/or restarting.

USB Device Tree
 USB 3.1 Bus:
  Host Controller Driver:	AppleT8103USBXHCI
   RTL2838UHIDIR:
     Product ID:	0x2838
     Vendor ID:	0x0bda  (Realtek Semiconductor Corp.)
     Version:	1.00
     Serial Number:	00000001
     Speed:	Up to 480 Mb/s
     Manufacturer:	Realtek
     Location ID:	0x00100000 / 1
     Current Available (mA):	500
     Current Required (mA):	500
     Extra Operating Current (mA):	0

**Steps To Reproduce**
1. Launch SDR++
2. Attempt to select RTL-SDR

**Screenshots**
[image]

AlexandreRouma, 2023-09-07:
Known issue, in the process of fixing it atm.
In the mean time you can use this specific build: https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/6091688808
I'll update this issue once the build is back in working order

o-Dan-o, 2023-09-07:
That worked. Thanks!!

AlexandreRouma, 2023-09-13:
Can you confirm the latest nightly works as expected now?

o-Dan-o, 2023-09-13:
Just installed "03:28:07, Sep 13 2023", RTL-SDR is now in the menu, did a quick test, app is working, no problems. Thanks!

closed 2023-09-13 by AlexandreRouma