AlexandreRouma/SDRPlusPlus #1187: unable to execute /usr/bin/sdrpp: No such file or directory

jonboy86, 2023-09-15:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Raspi 4
- RAM: Raspi 4
- GPU: Raspi 4
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
unable to execute /usr/bin/sdrpp: No such file or directory

**Steps To Reproduce**
1. ... installed as per instructions
2. ... running sdrpp from terminal gives me 
3. ... unable to execute /usr/bin/sdrpp: No such file or directory

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

AlexandreRouma, 2023-09-17:
Template isn't filled out properly but almost certainly downloded the deb for the wrong architecture.

closed 2023-09-17 by AlexandreRouma

kupian, 2023-12-25:
having this same issue also on an Rpi4. definitely downloaded the correct deb so this may need reopening.

AlexandreRouma, 2023-12-25:
no you didn't

kupian, 2023-12-25:
if it's not sdrpp_raspios_bullseye then that's my mistake. what should I be downloading?

AlexandreRouma, 2023-12-25:
the part you didn't mention says `armhf`, which means 32bit. There is no 64bit build and you have a 64bit os.

kupian, 2023-12-25:
thanks for the clarification. may be worth adding that to the readme somewhere as I've seen a lot of people missing that detail :)

prismspecs, 2024-04-23:
> the part you didn't mention says `armhf`, which means 32bit. There is no 64bit build and you have a 64bit os.

does this mean it must necessarily be built from source for 64bit Raspi OS?

AStoker, 2024-06-15:
@prismspecs , this just bit me 😂
Did you get it working?