AlexandreRouma/SDRPlusPlus #1189: Scanning from memory banks

krtz07, 2023-09-16:
**Feature description**
I'm a bit carious to know whether or not the roadmap has the scanning from frequencies within  memory banks feature planning to be implemented?

Thus far the SDR softwares which offers this has too many limitations either it's business related or too CPU consuming. It will be a great service to this community for this feature to be implemented on SDR++.

AlexandreRouma, 2023-09-16:
already planned

ItsMe-TJ, 2023-10-07:
> already planned

any ETA on when we might see it? No rush or anything, just incredibly exciting! It's the one thing that's keeping me on SDR#, but that's all I use it for as I much prefer SDR++, so having to switch back and forth is annoying.

craigerl, 2024-05-18:
agreed, killer feature - thumbs up