AlexandreRouma/SDRPlusPlus #1190: UHD / USRP Support DC removal

deeptho, 2023-09-16:
**Feature description**
The UHD/USRP code by default applies DC removal. This causes problems when using "center tuning" on AM 
stations. Basically it causes the carrier to be removed and demodulation to fail. 

So if I use center tuning and tune to 648kHz AM, I get poor audio. If I tune to 600 kHz first, then switch 
to "normal tuning" and tune 648kHz the audio becomes clear. This becomes tedious when switching to 
other bands and destroys the main advantage of center tuning.

The best solution is perhaps to add a gui option and allow the user a choice, 
but simpler is just to remove the default dc removal. This is a one-liner:
add 	
```_this->dev->set_rx_dc_offset(false, _this->chanId);```
in USRPSourceModule::start(void* ctx)

I do not know if this has any downsides, but it seems to work. Ettus claims that this default was chosen
for "greater stability" and recommends always tuning to a frequency such that the center frequency is not
in the middle of the pass band.
Therefore, another solution would be to make "center tuning mode" do that by applying a frequency offset. Note that is different from the one used in "offset mode" because we do not want frequency adjustment in this case. 
Choosing the proper offset needs to be done carefully
as the desired frequency needs to be in the current pass band, even at low bandwidth settings. It should also be 
larger than the bandwidth of the desired signal. 

So this sounds a bit complicated.

AlexandreRouma, 2023-09-16:
> I do not know if this has any downsides, but it seems to work. Ettus claims that this default was chosen
for "greater stability" and recommends always tuning to a frequency such that the center frequency is not
in the middle of the pass band.

You should never use the very center of the IQ unless it comes from DDC and is mathematically guaranteed there won't be anything weird in the center. This is not the case with all SDR hardware and in the case of the USRPs is something you have to work around. Even if you disable DC removal, you'll just get a spike instead which will also make your audio worse. Therefor, just disabling it is not a solution either even if it appears to work "better" (not to mention moving away from manufacturer defaults).

Adding a GUI option for this is a bad idea, SDR++ already has software based IQ correction for all sources so there would be two options in the same menu that do more or less same thing differently, very confusing for users.

> Therefore, another solution would be to make "center tuning mode" do that by applying a frequency offset. Note that is different from the one used in "offset mode" because we do not want frequency adjustment in this case.

Thereby going against the entire point of center tuning? This mode specifically there to lock the VFO and hardware to be identical...

To me the solution here is clear. Don't use center tuning and learn to avoid the center if it even ends up tuning there. You can easily drag the frequency scale to move the center frequency if needed.

deeptho, 2023-09-19:
You are right that it is not a good solution: I tried it now and the dc component is incorrect, creating problems 
on ssb.

Still, the current way of working is cumbersome of usrp. For instance, I use a low bandwidth and sampling rate in some cases to have maximum processing gain. When "Center tuning" works fine for tuning (e.g., airspy) you can easily navigate a band much wider than the bandwith, simply by clicking. E.g. clicking to the high end of the spectrum centers that frequency. You can then click another frequency and thus navigate the full band with fewer clicks (or by arrow keys)

When not using "center tuning" more clicks are needed in order to leave the band.
 
I am currently experimenting with setting an offset of 200khz for usrp,. This solution is not perfect, but for short and
medium wave it should be ok, as long as the bandwidth is above 400khz. 

Perhaps an option to set the offset by the user could be nice. Somehing like "off center tuning", for example
"always center device 200khz above the tuned frequency", with 200khz being the user option

AlexandreRouma, 2024-10-07:
Not sure why I left this open but since the DC offset correction is done by libuhd in software, there is no point having the option when SDR++ has its own DC removal code.

deeptho, 2024-10-07:
> Not sure why I left this open but since the DC offset correction is done by libuhd in software, there is no point having the option when SDR++ has its own DC removal code.

Maybe because the ticket is not about "needing DC removal" but about the negative effects of dc removal?

AlexandreRouma, 2024-10-07:
indeed it is, guess I shoudn't go through issues while being distracted