AlexandreRouma/SDRPlusPlus #1193: Frequency offset when using spyserver

helioshk, 2023-09-20:
When connecting to a spyserver, the signal frequencies are 6250 Hz too low, in the FFT as well as the audio, compared to other spyserver client (sdr#) and calibrated reference.

6250 Hz would be half the NFM bandwidth, but I don't know where that is coming from.

When connecting with sdrpp to a local Airspy, the frequency is accurate.

helioshk, 2023-09-29:
I think the problem is, that the frequency range can be changed in SDR++, even if spyserver doesn't allow changing the frequency:
```
# Allow clients to retune and change of gain of the device
#
allow_control = 0
```

AlexandreRouma, 2023-09-29:
Not a bug, SDR++ just doesn't support locking the frequency range.