AlexandreRouma/SDRPlusPlus #1200: Assertion failure in stl_vector

Frozen-Sherbet, 2023-09-29:
**Hardware**
- CPU: AMD Ryzen 5 7530U with Radeon Graphics (12) @ 4.546GHz
- RAM: 22897MiB
- GPU: AMD ATI
- SDR: rtl-sdr

**Software**
- Operating System: Arch Linux x86_64
- SDR++: v1.1.0 build date 09/28/2023

**Bug Description**
Process opens window, gets to Initializing Radio and core dumps

**Steps To Reproduce**
1. Run sdrpp

**Console Log**
[28/09/2023 20:17:20.000] [INFO] SDR++ v1.1.0
[28/09/2023 20:17:20.000] [INFO] Loading config
[28/09/2023 20:17:20.000] [INFO] Using OpenGL 3.0
[28/09/2023 20:17:20.000] [INFO] Loading icons
[28/09/2023 20:17:20.000] [INFO] Loading band plans
[28/09/2023 20:17:20.000] [INFO] Loading band plans color table
[28/09/2023 20:17:20.000] [INFO] Loading modules
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[28/09/2023 20:17:20.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[28/09/2023 20:17:20.000] [INFO] Initializing Audio Sink (audio_sink)
[28/09/2023 20:17:20.000] [INFO] Initializing File Source (file_source)
[28/09/2023 20:17:20.000] [INFO] Initializing Frequency Manager (frequency_manager)
[28/09/2023 20:17:20.000] [INFO] Initializing Network Sink (network_sink)
[28/09/2023 20:17:20.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[28/09/2023 20:17:20.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[28/09/2023 20:17:20.000] [INFO] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = RtAudio::DeviceInfo; _Alloc = std::allocator<RtAudio::DeviceInfo>; reference = RtAudio::DeviceInfo&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)

AlexandreRouma, 2023-09-29:
RtAudio bug, nothing I can do. Following these instructions will disable RtAudio's JACK backend and should allow SDR++ to run.
https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

closed 2023-09-29 by AlexandreRouma

Frozen-Sherbet, 2023-09-29:
Thank you for pointing me in the right direction.