AlexandreRouma/SDRPlusPlus #1204: Rtaudio causes program to crash on startup

Jomjom79, 2023-10-04:
Since yesterday sdr++ has started crashing on startup. 

The only way i can get it to start is to remove rtaudio.dll from the directory, then enter the program, set the sink to Network and then quit. Copy rtaudio back to the directory and then start the program and choose audio and what sink to use. If i set sink to none before quitting the program it will start up again. If i leave it on audio and any working sink (speakers/virtualcable) it will crash again the next time i start it up. 

Have reinstalled the program, sound drivers, virtual cable but nothing seem to work. 

Running sdrpp.exe - s shows this message 

C:\sdrpp_windows_x64 (2)\sdrpp_windows_x64>sdrpp -s
[04/10/2023 12:30:18.000] [INFO] SDR++ v1.1.0
[04/10/2023 12:30:18.000] [INFO] Loading config
[04/10/2023 12:30:18.000] [INFO] =====| SERVER MODE |=====
[04/10/2023 12:30:18.000] [INFO] Loading modules
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/airspyhf_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/airspy_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/audio_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/bladerf_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/file_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/hackrf_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/hermes_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/limesdr_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/perseus_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/plutosdr_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/rfspace_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/rtl_sdr_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/rtl_tcp_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/sdrplay_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/sdrpp_server_source.dll
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/soapy_source.dll
[04/10/2023 12:30:18.000] [ERROR] Couldn't load C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/soapy_source.dll. Error code: 126
[04/10/2023 12:30:18.000] [INFO] Loading C:/sdrpp_windows_x64 (2)/sdrpp_windows_x64/modules/spyserver_source.dll
[04/10/2023 12:30:18.000] [INFO] Initializing Airspy Source (airspy_source)
[04/10/2023 12:30:18.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[04/10/2023 12:30:18.000] [INFO] Initializing Audio Source (audio_source)
[04/10/2023 12:30:18.000] [INFO] Initializing BladeRF Source (bladerf_source)
[04/10/2023 12:30:18.000] [ERROR] Could not list devices -7
[04/10/2023 12:30:18.000] [INFO] Initializing File Source (file_source)
[04/10/2023 12:30:18.000] [INFO] Initializing HackRF Source (hackrf_source)
[04/10/2023 12:30:18.000] [INFO] Initializing Hermes Source (hermes_source)
[04/10/2023 12:30:18.000] [INFO] Initializing LimeSDR Source (limesdr_source)
[04/10/2023 12:30:18.000] [INFO] Initializing PerseusSDR Source (perseus_source)
perseus: perseus_init()
perseus: EXE: [C:\sdrpp_windows_x64 (2)\sdrpp_windows_x64\sdrpp.exe]
perseus: path: [C:\sdrpp_windows_x64 (2)\sdrpp_windows_x64\] name: [sdrpp.exe]
perseus: Found device with VID/PID 8086:9D2F on BUS1 ADDR0
perseus: Found device with VID/PID 8087:0A2A on BUS1 ADDR3
perseus: Found device with VID/PID 0BDA:2838 on BUS1 ADDR13
perseus: Found device with VID/PID 04F3:0235 on BUS1 ADDR7
perseus: Found device with VID/PID 0BDA:0129 on BUS1 ADDR4
[04/10/2023 12:30:18.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
[04/10/2023 12:30:18.000] [INFO] Initializing RFspace Source (rfspace_source)
[04/10/2023 12:30:18.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
Found Rafael Micro R820T tuner
[04/10/2023 12:30:19.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[04/10/2023 12:30:19.000] [INFO] Initializing SDR++ Server Source (sdrpp_server_source)
[04/10/2023 12:30:19.000] [INFO] Initializing SDRplay Source (sdrplay_source)
[04/10/2023 12:30:19.000] [ERROR] Could not intiatialized the SDRplay API. Make sure that the service is running.
[04/10/2023 12:30:19.000] [INFO] Initializing SpyServer Source (spyserver_source)
[04/10/2023 12:30:19.000] [INFO] Running post-init for Airspy Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for AirspyHF+ Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for Audio Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for BladeRF Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for File Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for HackRF Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for Hermes Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for LimeSDR Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for PerseusSDR Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for PlutoSDR Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for RFspace Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for RTL-SDR Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for RTL-TCP Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for SDR++ Server Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for SDRplay Source
[04/10/2023 12:30:19.000] [INFO] Running post-init for SpyServer Source
[04/10/2023 12:30:19.000] [INFO] RTLSDRSourceModule 'RTL-SDR Source': Menu Select!
[04/10/2023 12:30:19.000] [INFO] Ready, listening on 0.0.0.0:5259

All other programs seems to work fine. 

jomjom79

Jomjom79, 2023-10-04:
Well it works now. Tried running it in compatibility mode with windows 8. Now it starts no matter what audio sink im choosing. You have to wonder why this was not necessary yesterday? Running it in CMD yields the same crash still.. 

Windows is like a rebellious teenager, you never know when its going to act up.. I miss my Amiga. 

Jomjom79

AlexandreRouma, 2023-10-14:
In any case, as the issue template says `-s` is for the older 1.0.4 release so the logs you provided are a server startup which doesn't load any audio components. Since it seems to be resolved anyway I'll close this issue.

closed 2023-10-14 by AlexandreRouma