AlexandreRouma/SDRPlusPlus #1205: Key/mouse Shortcut for snap intervals *10 /10

THEKIWSA, 2023-10-04:
First, SDR++ is my goto when I just wanna listen to some ham.  The Issue: just like it says in the title,Most standalone SDR's (that I've looked at) have a couple buttons below the VFO knob for *10 /10 on the snap interval. Might I suggest arrowup/dn and maybe mouse4/5 (whatever the mouse thumb buttons are called in code nowadays the ones that default to fore and back in a browser.) This would help a lot when cruising between bands via mouse scroll, and fine tuning some crazy off channel stuff especially when spanning like ~3MHz on the waterfall. Thanks in advance for your consideration, and for all you guys' hard work thus far.

Please remember:
"Carpe Crustulorem" -- Seize the Doughnut

AlexandreRouma, 2023-10-11:
> arrowup/dn

Never use cursor keys for things that aren't explicitely moving things around. This is a UI/UX failure.

> mouse4/5

Not all mice have these, and no laptop does at all.

The only usable keys would be Shift and Alt. Ctrl is already used for displaying more VFO information.

AlexandreRouma, 2023-10-11:
Implemented, use Left Shift for 10x and Left Alt for 0.1x

closed 2023-10-11 by AlexandreRouma