AlexandreRouma/SDRPlusPlus #1206: Samplerate more than 20Mhz

johnsonZ72, 2023-10-11:
**Feature description**
Good day!
Plutosdr allows 61 MHz samplerate
In the program you can only select max-20MHz
It is possible to increase the samprate?

AlexandreRouma, 2023-10-11:
Wrong. The pluto's USB interface can barely handle 5MS/s, and 9 with firmware modifications. Even the current 20MS/s limit is overkill.

closed 2023-10-11 by AlexandreRouma

johnsonZ72, 2023-10-11:
I don't use USB, I use the network
Network usage at sample rate 20Mhz - MBit/s
Theoretically at 60Mhz - 900MBit/s

AlexandreRouma, 2023-10-11:
You're not using a plutosdr then. You're using one of those aliexpress clones which has completely different hardware. Being accurate in your issue description is important.
In any case, even on gigabit ethernet, you will not get 61MS/s and the widest officially supported bandwidth is 20MHz. 
https://wiki.analog.com/university/tools/pluto/devs/specs

johnsonZ72, 2023-10-11:
Sorry for misleading you about the SDR receiver
I am using a development board with AD9361 chip and gigabit interface.
Here is a link to the performance of chips with firmware 0.32
https://wiki.analog.com/university/tools/pluto/users/customizing#updating_to_the_ad9364

In the gnuradio and io-oscilloscope I get a 60 MHz samlrate.
Is it possible to increase the sample rate in SDR++?