AlexandreRouma/SDRPlusPlus #1209: Notch Filter

Sultan-papagani, 2023-10-14:
a notch filter with adjustable frequency
so we don't have to listen to constant humming

[image]

[image]

JerryXu09, 2025-04-19:
[image]
I think a notch filter would be very useful ❤️

Sultan-papagani, 2025-04-20:
It's been two years.

he's had over 730 days to add a basic bandstop filter.

I've moved on to other sdr softwares myself, no need to keep waiting for this one.

AlexandreRouma, 2025-04-20:
Pay me if you want it quicker then

FlashlightET, 2025-11-16:
you are overworking yourself with your contribution model (or rather, lack thereof)