AlexandreRouma/SDRPlusPlus #1211: Segmantation Fault

naterade1776, 2023-10-16:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel Celeron N2840 (2) @ 2.582GHz
- RAM: 3842MiB
- GPU: Intel Atom Processor Z36xxx/Z37xxx Series
- SDR: Local RTL-SDR Dongle V3.

**Software**
- Operating System: Alpine Linux v3.18 x86_64
- SDR++: N/A - Compiled October 15, 2023

**Bug Description**
~$sdrpp
[16/10/2023 15:08:16.000] [INFO] SDR++ v1.1.0
[16/10/2023 15:08:16.000] [INFO] Loading config
[16/10/2023 15:08:17.000] [INFO] Using OpenGL 3.0
[16/10/2023 15:08:17.000] [INFO] Loading icons
[16/10/2023 15:08:17.000] [INFO] Loading band plans
[16/10/2023 15:08:17.000] [INFO] Loading band plans color table
[16/10/2023 15:08:17.000] [ERROR] Menu element is missing name key
[16/10/2023 15:08:17.000] [INFO] Loading modules
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[16/10/2023 15:08:17.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[16/10/2023 15:08:17.000] [INFO] Initializing Airspy Source (airspy_source)
[16/10/2023 15:08:17.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[16/10/2023 15:08:17.000] [INFO] Initializing Audio Sink (audio_sink)
[16/10/2023 15:08:17.000] [INFO] Initializing Audio Source (audio_source)
ALSA lib pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

ALSA lib pcm_dsnoop.c:566:(snd_pcm_dsnoop_open) unable to open slave

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

Segmentation fault

**Steps To Reproduce**
I compiled the program from source on Alpine Linux. I have tried launching with a desktop file and from command line.
Issue is unchanged when invoked with command flags.

sizzlik, 2023-11-03:
seems like the same problem we have here https://github.com/AlexandreRouma/SDRPlusPlus/issues/1090

AlexandreRouma, 2023-11-08:
Not a bug in SDR++, audio is just misconfigured on your system.
And @sizzlik This literally has nothing to do with issue number 1090...

closed 2023-11-08 by AlexandreRouma