AlexandreRouma/SDRPlusPlus #1224: no way to add airspy server ui is not working

ta3pks, 2023-11-10:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: apple m1
- RAM: 8gb
- GPU: N/A
- SDR: airspy remote

**Software**
- Operating System: macos 14.1
- SDR++: 1.1.0 02:25:47 oct 29 2023
**Bug Description**
From the manual what i understand is that source section should contain 2 inputs to enter host and port after selecting Airspy.
However in my case theres just two empty dropdowns in which i cannot input anything effectively rendering the entire app unusable.

**Steps To Reproduce**
1. go to applications folder and click on sdr++.app
2. select Airspy in the source menu
3. no way to enter the server details

**Screenshots**
[image]
[image]
[image]

AlexandreRouma, 2023-11-10:
"SpyServer" and "Airspy" are two different things, shocking I know...
What you want is "SpyServer", Airspy is the device.

closed 2023-11-10 by AlexandreRouma

ta3pks, 2023-11-10:
oh i see thanks