AlexandreRouma/SDRPlusPlus #1225: compile fix & missing package

folkertvanheusden, 2023-11-10:
core/src/utils/networking.cpp requires #include <stdexcept>
and the readme fails to mention that for ubuntu one needs to install librtlsdr-dev

AlexandreRouma, 2023-11-10:
1) You're wrong, the CI checks every commit and it builds correctly on all supported operating systems, including ubuntu.
2) No shit sherlock, you need to install the driver for your SDR???? Who could have thought?????? And yes, it is absolutely mentioned in the readme:
[image]

closed 2023-11-10 by AlexandreRouma

folkertvanheusden, 2023-11-10:
I'm talking about Ubuntu 23.10 here (may not be officially supported yet, but then it is a fix for the future). And it was not mentioned in the part where it tells you how to prepare your debian/ubuntu system (the apt-get part).

AlexandreRouma, 2023-11-10:
> I'm talking about Ubuntu 23.10 here (may not be officially supported yet, but then it is a fix for the future).

No, it's known to work, there's just not a CI build for it yet. What you said here "core/src/utils/networking.cpp requires #include <stdexcept>" is just plain wrong, literally line 4 of the file:
[image]

I don't know how old of a commit you're using but it must be at least a year old...

folkertvanheusden, 2023-11-10:
> No, it's known to work, there's just not a CI build for it yet. What you said here "core/src/utils/networking.cpp requires #include <stdexcept>" is just plain wrong, literally line 4 of the file: [...]

Ah I was using the 1.0.4 release.

Indeed the latest version on github compiles & runs fine.