AlexandreRouma/SDRPlusPlus #1231: Allow Left and Right Channel for Audio

grahampaull, 2023-11-17:
The ability to select which speaker a VFO plays through.

I currently listen to 2 VFOs but would like to set an Audio Sink to play through the Left Speaker/Channel and another Audio Sink to play through the Right Speaker/Channel.

I also use VoxCall which allows me to feed audio to my rdio-scanner app, it allows to feed stereo, left or right. 

This works well using the the AUX-VFO plugin on SDR#:
[image]

grahampaull, 2023-11-19:
> Already implemented in the `new_sinks` branch but it's not currently usable. Will notify when merged back.

Excellent! Thank you.

AlexandreRouma, 2024-02-15:
Just realized this is a duplicate of #653, follow that issue instead

closed 2024-02-15 by AlexandreRouma