AlexandreRouma/SDRPlusPlus #1233: Crash when removing additional VFO

benchun, 2023-11-18:
**Hardware**
- CPU: 2.4 GHz 8-Core Intel Core i9
- RAM: 64 GB 2667 MHz DDR4
- GPU: AMD Radeon Pro 5500M 8 GB
- SDR: RTL-SDR v3 (local)

**Software**
- Operating System: MacOS 14.1.1 (23B81)
- SDR++: v1.1.0 (Built at 2:24:47, Oct 29 2023)

**Bug Description**
Created 3 additional VFOs named "Radio 2", "Radio 3", and "Radio 4". Decided I did not want these additional VFOs. Tried to delete one of them and SDR++ crashes. This behavior is repeatable and I can not delete any of these additional VFOs.

**Steps To Reproduce**
1. Create additional VFO of type radio.
2. Click minus button next to new VFO.
3. At the confirmation dialog like "Deleting Radio 2. Are you sure?" click Yes
4. SDR++ crashes

**Additional info**
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               sdrpp [41136]
Path:                  /private/var/folders/*/SDR++.app/Contents/MacOS/sdrpp
Identifier:            oorg.sdrpp.sdrpp
Version:               1.1.0
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-11-17 20:45:31.3519 -0800
OS Version:            macOS 14.1.1 (23B81)
Report Version:        12
Bridge OS Version:     8.1 (21P1069)
Anonymous UUID:        0C1F8822-F547-7C0C-23CF-74840BC145FA

Sleep/Wake UUID:       D9036190-1D05-43FC-A2E9-0EC660EE22FB

Time Awake Since Boot: 350000 seconds
Time Since Wake:       14178 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   sdrpp [41136]

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff81185f7ce __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff811897f30 pthread_kill + 262
2   libsystem_c.dylib             	    0x7ff8117b6a49 abort + 126
3   libc++abi.dylib               	    0x7ff811850c72 abort_message + 241
4   libc++abi.dylib               	    0x7ff811842d3f demangling_terminate_handler() + 47
5   libc++abi.dylib               	    0x7ff8118500cb std::__terminate(void (*)()) + 6
6   libc++abi.dylib               	    0x7ff811850070 std::terminate() + 32
7   radio.dylib                   	       0x10874e224 dsp::block::~block() + 180
8   radio.dylib                   	       0x10876cc48 RadioModule::~RadioModule() + 424
9   radio.dylib                   	       0x108749fe3 _DELETE_INSTANCE_ + 19
10  libsdrpp_core.dylib           	       0x1073ac52a ModuleManager::deleteInstance(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) + 234
11  libsdrpp_core.dylib           	       0x1072c9827 module_manager_menu::draw(void*) + 1447
12  libsdrpp_core.dylib           	       0x1072f2924 Menu::draw(bool) + 2308
13  libsdrpp_core.dylib           	       0x1072c3ad1 MainWindow::draw() + 4033
14  libsdrpp_core.dylib           	       0x1073d5509 backend::renderLoop() + 1305
15  libsdrpp_core.dylib           	       0x107280790 sdrpp_main(int, char**) + 20272
16  dyld                          	    0x7ff81150f3a6 start + 1942

AlexandreRouma, 2024-01-27:
Can you confirm this still occurs with the latest nightly?

AlexandreRouma, 2024-01-29:
I'm pretty sure the root cause is the same as #816, follow that issue one instead.

closed 2024-01-29 by AlexandreRouma