AlexandreRouma/SDRPlusPlus #1234: Build Failure on openSUSE Tumbleweed - gcc with "-Werror=return-type"

mnhauke, 2023-11-21:
master fails to build on openSUSE Tumbleweed with the distro provided CFLAGS and gcc 13.2.1 when building on OBS.

Compiling with "-Werror=return-type" results in
```
[    9s] /home/abuild/rpmbuild/BUILD/SDRPlusPlus-1.0.4+git20231029/core/src/utils/proto/http.cpp: In member function 'int net::http::Client::recvRequestHeader(net::http::RequestHeader&, int)'
[    9s] /home/abuild/rpmbuild/BUILD/SDRPlusPlus-1.0.4+git20231029/core/src/utils/proto/http.cpp:260:5: error: control reaches end of non-void function [-Werror=return-type]
...
[    9s] /home/abuild/rpmbuild/BUILD/SDRPlusPlus-1.0.4+git20231029/core/src/utils/proto/http.cpp: In member function 'int net::http::Client::recvResponseHeader(net::http::ResponseHeader&, int)':
[    9s] /home/abuild/rpmbuild/BUILD/SDRPlusPlus-1.0.4+git20231029/core/src/utils/proto/http.cpp:277:5: error: control reaches end of non-void function [-Werror=return-type]
...
[   11s] /home/abuild/rpmbuild/BUILD/SDRPlusPlus-1.0.4+git20231029/core/src/signal_path/../dsp/chain.h: In member function 'dsp::Processor<D, D>* dsp::chain<T>::blockBefore(dsp::Processor<D, D>*) [with T = dsp::complex_t]':
[   11s] /home/abuild/rpmbuild/BUILD/SDRPlusPlus-1.0.4+git20231029/core/src/signal_path/../dsp/chain.h:170:9: error: control reaches end of non-void function [-Werror=return-type]
...
[   22s] /home/abuild/rpmbuild/BUILD/SDRPlusPlus-1.0.4+git20231029/core/src/dsp/demod/ssb.h: In member function 'double dsp::demod::SSB<T>::getTranslation() [with T = dsp::stereo_t]':
[   22s] /home/abuild/rpmbuild/BUILD/SDRPlusPlus-1.0.4+git20231029/core/src/dsp/demod/ssb.h:116:9: error: control reaches end of non-void function [-Werror=return-type]
```
CFLAGS
```
[    2s] + /usr/bin/cmake /home/abuild/rpmbuild/BUILD/SDRPlusPlus-1.0.4+git20231029/. '-GUnix Makefiles' -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=/var/lib -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=var -DCMAKE_INSTALL_RUNSTATEDIR:PATH=run -DCMAKE_INSTALL_LIBDIR:PATH=lib64 -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DCMAKE_BUILD_TYPE=RelWithDebInfo '-DCMAKE_C_FLAGS=-O2 -Wall -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g' '-DCMAKE_CXX_FLAGS=-O2 -Wall -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g' '-DCMAKE_Fortran_FLAGS=-O2 -Wall -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g' '-DCMAKE_EXE_LINKER_FLAGS=-flto=auto -Wl,--as-needed -Wl,-z,now' '-DCMAKE_MODULE_LINKER_FLAGS=-flto=auto -Wl,--as-needed' '-DCMAKE_SHARED_LINKER_FLAGS=-flto=auto -Wl,--as-needed -Wl,-z,now' -DLIB_SUFFIX=64 -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DBUILD_SHARED_LIBS:BOOL=ON -DBUILD_STATIC_LIBS:BOOL=OFF -DCMAKE_COLOR_MAKEFILE:BOOL=OFF -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_MODULES_INSTALL_DIR=/usr/lib64/cmake/sdrpp-git -DCMAKE_SHARED_LINKER_FLAGS= -DUSE_INTERNAL_LIBCORRECT=OFF -DOPT_BUILD_AIRSPYHF_SOURCE=ON -DOPT_BUILD_LIMESDR_SOURCE=ON -DOPT_BUILD_BLADERF_SOURCE=ON -DOPT_BUILD_PLUTOSDR_SOURCE=ON -DOPT_BUILD_USRP_SOURCE=ON -DOPT_BUILD_WEATHER_SAT_DECODER=OFF -DOPT_BUILD_SPYSERVER_SOURCE=ON
```

Workaround/Fix
https://build.opensuse.org/package/view_file/home:mnhauke:sdr-devel/sdrpp-git/fix-build.patch?expand=1

AlexandreRouma, 2023-11-21:
Will fix tomorrow but tell the maintainer that the version is wrong, it should say 1.1.0 not 1.0.4

mnhauke, 2023-11-22:
> Will fix tomorrow but tell the maintainer that the version is wrong, it should say 1.1.0 not 1.0.4

Thanks, I just updated the version string for the package.

closed 2024-01-22 by AlexandreRouma