AlexandreRouma/SDRPlusPlus #1236: how to install on ubuntu 23.10, getting

kocoman1, 2023-11-27:
how to install on ubuntu 23.10, getting 

Package libvolk2-dev is not available, but is referred to by another package.
--
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
libvolk-dev libcpu-features-dev

but the sdrpp_ubuntu_jammy_amd64.deb package give error

sdrpp depends on libvolk2-dev; however:
Package libvolk2-dev is not installed.

thx

michaelself10, 2023-12-08:
> > This is how I install SDR++ on Ubunut 22.04 - jammy.
> > 
> > 1. `sudo apt install libfftw3-dev libglfw3-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev`
> > 2. Download the release `sdrpp_ubuntu_jammy_amd64.deb` from the releases github page
> > 3. Try to install the release from step 2 by running `sudo dpkg -i sdrpp_ubuntu_jammy_amd64.deb`. You will have an error of dependencies that require you to install `libzstd-dev`.
> > 4. Install `libzstd-dev` by running `sudo apt install libzstd-dev`
> > 5. Finally run again `sudo dpkg -i sdrpp_ubuntu_jammy_amd64.deb` to install SDR++
> > 
> > After this, you should have SDR version 1.1.0 running perfectly.
> 
> I am having the same problem as the OP, so I tried what you wrote. It still doesn't have libvolk2-dev so it still doesnt work

I just solved the problem by following @geogarden instructions. By not needing libvolk2, it works! Thank you!

samysousa, 2023-12-30:
> I just solved the problem by following @geogarden instructions. By not needing libvolk2, it works! Thank you!

What instructions ? i only see a reference to "did something don't remember". 
Trying to also get this updated, Libvolk3 is now the main package on 23.10 , and it just does not want to live with the latest update of SDRPP. SDRPP still works but once you Fix packages/Remove SDRPP, you can't reinstall because of the libvolk2-dev error.

AlexandreRouma, 2024-01-22:
There is no deb available for this distro yet, only way is from source at the moment, check back the nightly regularly to see if one is available.

closed 2024-01-22 by AlexandreRouma

AlexandreRouma, 2024-01-25:
a deb is now available.