AlexandreRouma/SDRPlusPlus #1240: LibreSDR support

gm8arv, 2023-12-01:
Any chance of adding support for the LibreSDR - a Pluto++ variant as sold here:

  https://www.aliexpress.com/item/1005005165734650.html

It has two RX and two TX channels and it's already supported by Simon's SDR Console and Alberto's Satsagen.  I don't have a pointer to the API, I'm afraid.

Thanks,
David

AlexandreRouma, 2023-12-01:
If it's pluto compatible then use the pluto source.
I don't plan on supporting goofy clones with no real company backing them up. If they don't have their own API and I don't have a unit, it's not getting support.

gm8arv, 2023-12-01:
Unfortunately the Pluto version doesn't work.  As I mentioned, SDR Console and Satsagen do support this unit and I suspect it's the same as the Pluto++ which is also in use.  Thanks for looking, but perhaps someone else might take this up.  Cheers, David

gm8arv, 2023-12-02:
I have some information on the APIs:

1.  All Pluto-derived devices, like ADALM-PLUTO, Pluto+, LibreSDR, AntSDR e310, AntSDR e200, and so on, can be accessed by the same APIs from the Libiio library.  The slight hardware differences of the above models are managed by the custom firmware shipped by the vendors.

2. The Libiio alternative is the SoapySDR library. Soapy SDR is a vendor and platform-neutral SDR support library,

I appreciate that you don't have a device (I've been in the same situation) but I would be happy to test for you.