AlexandreRouma/SDRPlusPlus #1250: Unable to build on macOS Sonoma 14.1.1, Apple Silicon M2

girishadurrel, 2023-12-13:
I get the following error when I try to build 

```
SDRPlusPlus/source_modules/audio_source/src/main.cpp:11:10: fatal error: 'RtAudio.h' file not found
#include <RtAudio.h>
```

I followed this [guide](https://www.thetechprepper.com/sdr/installing-sdr-on-macos-monterey) with few modifications. 

`brew install cmake volk glew glfw airspy airspyhf hackrf rtl-sdr soapysdr portaudio pandoc pkg-config codec2`
`brew install rtaudio@5.2.0` (modification from the above reference)
`$ sudo ln -s /opt/homebrew/include /usr/local/include`
`$ sudo ln -s /opt/homebrew/lib /usr/local/lib`
`git clone https://github.com/AlexandreRouma/SDRPlusPlus.git`
`cd SDRPlusPlus`
`mkdir build && cd build`
```
cmake .. \
  -DOPT_BUILD_AUDIO_SINK=OFF \
  -DOPT_BUILD_BLADERF_SOURCE=OFF \
  -DOPT_BUILD_M17_DECODER=OFF \
  -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON \
  -DOPT_BUILD_PLUTOSDR_SOURCE=OFF \
  -DOPT_BUILD_PORTAUDIO_SINK=ON \
  -DOPT_BUILD_SOAPY_SOURCE=ON \
  -DOPT_BUILD_SDRPLAY_SOURCE=OFF  (modification from the above reference)
```
` make -j$(sysctl -n hw.ncpu)` (causes failure)

girishadurrel, 2023-12-13:
on `/usr/local/include` I do see the following

```
$ ls -l rtaudio*
lrwxr-xr-x  1 girisha  admin  45 Dec  8 21:51 rtaudio -> ../Cellar/rtaudio@5.2.0/5.2.0/include/rtaudio
```

AlexandreRouma, 2023-12-13:
Well obviously following some random unofficial build guide doesn't work... Maybe follow the one in the readme instead?
https://github.com/AlexandreRouma/SDRPlusPlus#building-on-macos
Delete everything and reclone the repo before doing so.

girishadurrel, 2023-12-16:
Thank you for your response. I cleaned everything up and then followed the MacOS guide, I am getting the following error now

```
/Users/girisha/Projects/SDRPlusPlus/source_modules/plutosdr_source/src/main.cpp:8:10: fatal error: 'iio.h' file not found
#include <iio.h>
```

girishadurrel, 2023-12-17:
I tried installing libiio
```
  $  curl -LJO https://raw.githubusercontent.com/tfcollins/homebrew-formulae/master/libiio.rb
  $  brew install --build-from-source ./libiio.rb
 ```
  still says the same error

girishadurrel, 2023-12-17:
then saw your comment [here](https://github.com/AlexandreRouma/SDRPlusPlus/issues/119#issuecomment-837081825), and that worked
all I did was to add this option in my cmake `-DOPT_BUILD_PLUTOSDR_SOURCE=OFF`

closed 2023-12-17 by girishadurrel