AlexandreRouma/SDRPlusPlus #1253: baseband filename template

rs1729, 2023-12-16:
Recording baseband, filename template `$r` would put "Unknown" in the filename:
https://github.com/AlexandreRouma/SDRPlusPlus/blob/794d6ff5acf66d5da7fb7fceba7b952c95305960/misc_modules/recorder/src/main.cpp#L470
Can baseband be something else than IQ data? If not, perhaps "IQ" would be the better choice
```
        const char* modeStr = "IQ";
```
cf.
https://github.com/AlexandreRouma/SDRPlusPlus/pull/1166#issuecomment-1708037722

AlexandreRouma, 2023-12-16:
The point of naming it "unknown" is because the `r`  template represents the modulation of the recorded signal, and since it's recording baseband, that modulation is 'unknown'. Having "IQ" in the name would be redundant with the "baseband" recording type

rs1729, 2023-12-16:
Yes, using `$t_$y$M$d_$h$m$s_$f_$r`it would be redundant. Using `$r`I wouldn't need `$t` anymore. When I choose "audio" it puts the (de)modulation in the filename. For baseband it puts "unknown". And if the "unknown" string would be set to an empty string for baseband, the filename would be something like "baseband_20231216_060000_400000000Hz_.wav", still including the "_" at the end.
With `$y$M$d_$h$m$s_$f_$r`a filename like "20231216_060000_400000000Hz_IQ.wav" would tell me, it is baseband. No modulation, just IQ data. The "audio" recordings with chosen demodulation would have different names.

AlexandreRouma, 2024-10-07:
Changed it such that baseband recording show IQ instead of unknown but audio recordings from a third party module will still shown "Unknown".

closed 2024-10-07 by AlexandreRouma