AlexandreRouma/SDRPlusPlus #1254: Doesn't work on android.

Tomrm96, 2023-12-17:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

Samsung galaxy tab s9 ultra
Rtl-sdr 4 

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

Latest nightly build 
Latest oneui/android 

**Bug Description**
A clear description of the bug.

The software detects the device and displays noise, but there are no waveforms just noise. I've plugged this into my laptop to see if there's an issue with the dongle all works fine, I downloaded sdr touch and it works fine on there. I've changed all the settings I can think off but nothing changes. 

**Steps To Reproduce**
1. ...
2. ...
3. ...

Plug in Rtl-sdr 4 

Click rtl sdr dongle 

Hit play, 

Search for known fm radio stations

Nothing appears just noise 

The same two spikes that look like radio signals are there wherever you go, which ever frequency you move too they are there but produce nothing. [image]

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

Tomrm96, 2023-12-17:
I was having some issues with sat dump and posted to the sat dump git, they resolved it by adding Rtl-sdr 4 support, not sure if this is the issue or not. As it seems sdr touch, the sdr driver has recently added support for 4.

AlexandreRouma, 2023-12-17:
not a bug just not currently supported. It will be announced once support is added.

closed 2023-12-17 by AlexandreRouma

Tomrm96, 2023-12-17:
Thanks,

Do you have any timelines?

AlexandreRouma, 2023-12-17:
no timeline, I'm quite busy

Tomrm96, 2023-12-17:
Fair enough, your work is appreciated have a good Christmas.