AlexandreRouma/SDRPlusPlus #1255: SDR++ on MS Windows 10 on a Asus VivoBook working with a RTLSDR v4 USBstick

ZephyrFanBoy, 2023-12-21:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- Asus VivoBook X513EAN_K513EA
- CPU:  11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz   2.80 GHz
- RAM: 16.0GB
- GPU:  intel GPU onboard
- SDR:  RTLSDR v4 USBstick (Local)  (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Windows 11 x64 22H2  22621.2861
- SDR++: sdrpp_windows_x64.zip  (SDR++ Nightly Build (currently 1.1.0) [AlexandreRouma](https://github.com/AlexandreRouma) released this Nov 22, 2022)

**Bug Description**
When launching the SDRcpp from file explorer nothing happens (seems to crash directly without any user-interaction of any kind.)

**Steps To Reproduce**
1. ... see *Bug Description*
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).
[image]

**Additional info**
Some better detailed tutorial and/or documentation would be nice for real SDR noobs.

ZephyrFanBoy, 2023-12-21:
Replacing the rtlsdr.dll (like to hardware vendor of the rtlsdr blog v4 usb stick advise) does not work.

ZephyrFanBoy, 2023-12-21:
> Replacing the rtlsdr.dll (like to hardware vendor of the rtlsdr blog v4 usb stick advise) does not work.

Restarting the PC seems to do the trick??

closed 2023-12-21 by ZephyrFanBoy

Mr-Precise, 2023-12-21:
Not enough write permissions?
Run as admin or move sdrpp to a user/desktop directory...