AlexandreRouma/SDRPlusPlus #1258: How do I change the recording output in SDR++ Android and why is screen recording with internal audio not possible in the Android version?

tteletubby, 2023-12-23:
I've recorded a NOAA WX Signal using the SDR++ Audio recorder just for it to say its on %ROOT%/Recordings which I can't find on my phone at all.

Then i've thought of just screen recording it and decoding it afterwards from the recording and seeing that audio isn't being recorded at all. What can I do?

tteletubby, 2023-12-24:
since it took me almost 2 days of searching...
answer is

**/storage/emulated/0/Downloads**

closed 2024-01-03 by tteletubby

zhaosge, 2024-01-14:
Differently, I find the "recordings" folder in /data/user/0.org.sdrpp.sdrpp/files/recordings...

Jasonzyt, 2025-07-29:
> Differently, I find the "recordings" folder in /data/user/0.org.sdrpp.sdrpp/files/recordings...

So I need to root my phone to access this directory...?