AlexandreRouma/SDRPlusPlus #1260: Incorrect equation to get samples from RTL-SDR

JVital2013, 2023-12-26:
**Hardware**
- CPU: Intel Core i5-6600K
- RAM: 16GB
- GPU: Radeon RX 580
- SDR: Local Nooelec NESDR SMArTee XTR SDR, and Nooelec NESDR SMArt v5 SDR (should be all RTL-SDR based dongles)

**Software**
- Operating System: Ubuntu 22.04 (or any OS; bug in source)
- Built against 794d6ff

**Bug Description**
In the following files:

- `source_modules/rtl_sdr_source/src/main.cpp`
- `source_modules/rtl_tcp_source/src/rtl_tcp_client.cpp`

The equation to take samples from the dongle and write them to `stream->writeBuf` is incorrect. The correct equation is (paraphrasing variable names here): 

```cpp
for (int i = 0; i < sampCount; i++) {
    _this->stream.writeBuf[i].re = (float)(buf[i * 2] - 127.4f) / 128.0f;
    _this->stream.writeBuf[i].im = (float)(buf[(i * 2) + 1] - 127.4f) / 128.0f;
}
```

The difference is the `127.4f` part, which you have as 127 or 128 depending on the source file. The incorrect equation results in a DC spike that is larger than it should be. Here are my sources for this change:

- https://cgit.osmocom.org/gr-osmosdr/commit/?id=0edfcfcba0ddf0fd5002dab70331d75a376bcf3f
- https://cgit.osmocom.org/gr-osmosdr/tree/lib/rtl/rtl_source_c.cc#n177
- https://github.com/pietern/goestools/blob/865e5c73683f8f688d102748e6b495ab5978768f/src/goesrecv/rtlsdr_source.cc#L164 (he refactored the equation strangely, but the math still works out the same)

SatDump has also been changed to this equation to a good effect.

**Steps To Reproduce**
1. Use any RTL-SDR and observe the DC spike
2. Make the suggested code change, and see the spike has gone away/been refuced, without using IQ correction

**Before**
[image]

**After**
[image]

AlexandreRouma, 2023-12-26:
This is not as clear cut as you make it out to be. I have collected the data of offset vs amplitude for 7 RTL-SDR dongles (3x RTL-SDRblog v4, 2x RTL-SDRblog v3, NESDRsmart and NESDRsmart XTR). This reveals that the offset is dependent on the amplitude for most modern dongles and can be quite random for older ones. Below is the raw data table and a graph representing the offset (from 127) as a function of IQ amplitude.

[image]
[image]

Calling the equation used "wrong" is rather silly. This equation has been traditionally used in most other RTL-SDR supporting software because that's just how uint8 samples are encoded.

AlexandreRouma, 2023-12-27:
After some more statistical testing, it seems the offset is indeed mostly 0.4
[image]

I'll implement the modification

closed 2023-12-27 by AlexandreRouma