AlexandreRouma/SDRPlusPlus #1264: Failed to Execute Child Process

Tomrm96, 2023-12-28:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**

Raspberry Pi 5
RTLSDR v4

**Software**

RaspberryPi OS 64 bit 
Nightly build on the release page: 
**Bug Description**
Unable to open, downloaded the latest from the releases page and I get the following error:

[image]

**Steps To Reproduce**

Download, extract, install. Try to run.

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

AlexandreRouma, 2023-12-28:
1) Filename:
[image]
2) Manual: 
[image]
3) Previous issues:
https://github.com/AlexandreRouma/SDRPlusPlus/issues/1187

Please think and do a bit of research before opening an issue.

closed 2023-12-28 by AlexandreRouma

Tomrm96, 2023-12-28:
Wow, your rude. Don't 64 bit OS's support 32 bit apps?

Conterganer, 2024-07-22:
And where can I find the version for the Pi5?