AlexandreRouma/SDRPlusPlus #1265: crash on Macbook on install / startup Intel Darwin crash report included

s-petersen, 2023-12-30:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
A clear description of the bug.

**Steps To Reproduce**
1. ... just tried to install / run
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.
Crash report includes all info Process:               sdrpp [706]
Path:                  /Volumes/VOLUME/*/SDR++.app/Contents/MacOS/sdrpp
Identifier:            oorg.sdrpp.sdrpp
Version:               1.1.0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           sdrpp [706]
User ID:               501

Date/Time:             2023-12-29 20:40:45.489 -0500
OS Version:            macOS 11.7.10 (20G1427)
Report Version:        12
Anonymous UUID:        6833672A-C651-4345-53F1-479A10113C22

Time Awake Since Boot: 430 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
sdrpp(706,0x11c22ee00) malloc: Heap corruption detected, free list is damaged at 0x600000b24c30
*** Incorrect guard value: 3420041017602237796
 

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff2031b90e __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff2034a5bd pthread_kill + 263
2   libsystem_c.dylib             	0x00007fff2029f406 abort + 125
3   libsystem_malloc.dylib        	0x00007fff2017f165 malloc_vreport + 548
4   libsystem_malloc.dylib        	0x00007fff201937e9 malloc_zone_error + 183
[... 398 lines cut ...]
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       501.3M       16 
__OBJC_RO                         70.3M        1 
__OBJC_RW                         2496K        2 
__TEXT                           204.5M      296 
__UNICODE                          588K        1 
mapped file                      223.9M       17 
shared memory                      784K       19 
===========                     =======  ======= 
TOTAL                              2.0G     1032 
TOTAL, minus reserved VM space     1.3G     1032 

Model: MacBookPro11,2, BootROM 478.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2.2 GHz, 16 GB, SMC 2.18f15
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.11)
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

s-petersen, 2023-12-30:
I was able to get around the issue by examine app contents, navigated to sdrpp and made an alias copied to desktop, changed icon, and it works fine now

AlexandreRouma, 2024-01-17:
Surprisingly can indeed replicate on my MacOS 10.15 2009 mac pro.

It seems to only happen when double clicking the bundle and not when running the executable contained within from the command line. It must be something related to the environment changing but I'm not sure...

s-petersen, 2024-01-17:
Yes it is strange, I built a new link to the executable, sent it to the desktop, and gave it the original icon, and the workaround seems to work fine

s-petersen, 2024-01-18:
Forgot to add, navigating through finder, and double clicking on the executable also runs, so maybe an issue in the package configuration relating to paths or something?, I'm not sure how a package is configured (I know nothing!), so just a guess.

AlexandreRouma, 2024-01-29:
@s-petersen Can you replicate this:

1) Download the latest nightly
2) Double-click it, it should **crash**
3) open a terminal and run `xattr -d com.apple.quarantine <sdr++ bundle file>`
4) Double-click it, it should **run**

s-petersen, 2024-01-29:
I copied and pasted into terminal and got
-bash: syntax error near unexpected token `newline'

AlexandreRouma, 2024-01-29:
you need to edit the command with the filename of the bundle for example `xattr -d com.apple.quarantine SDR++.app`

s-petersen, 2024-01-29:
Took a while for me to figure how to open a terminal in the correct folder, but that fixed it. Thanks!

AlexandreRouma, 2024-01-29:
Alright so you can confirm the program opens properly when double clicking after this "fix"?

s-petersen, 2024-01-29:
Yes. I first had to open the "untrusted" file the first time, then I applied the "fix" and now it works. Do you want me to try from scratch with a fresh download, and "fix" it before "trusting"it

AlexandreRouma, 2024-01-30:
yeah you can try that too

s-petersen, 2024-01-30:
That seems to have taken care of it, works fine. Thanks!

AlexandreRouma, 2024-01-30:
Alright.
Well it's not really a good fix, I still need to figure out the root cause of this so people don't have to do this.

tomduca, 2024-04-11:
> Alright. Well it's not really a good fix, I still need to figure out the root cause of this so people don't have to do this.

I get this when trying to launch from inside the package in case that it helps:
_“sdrpp” cannot be opened because it is from an unidentified developer._ 
[image]

And this when double clicking the downloaded package:
[image]