AlexandreRouma/SDRPlusPlus #1276: Can you run SDR++ in server mode from Android?

b4545, 2024-01-12:
### Discussed in https://github.com/AlexandreRouma/SDRPlusPlus/discussions/1275

<div type='discussions-op-text'>

<sup>Originally posted by **b4545** January 12, 2024</sup>
Hello, I'm running an RTL-SDR USB on a powered hub plugged into an old Android phone. I want to use SDR++ in server mode, but I could not find any way to do this. I'm aware of the RTL_TCP etc. methods, but I read that SDR++ server is more performant, which is why I'm trying to make it work. Also there is no root access on the phone. Thanks for any advice!</div>

AlexandreRouma, 2024-01-12:
Why would you ever want to run an SDR server on a *phone*?

b4545, 2024-01-12:
Hey thanks for getting back to me! Using an old phone is cost effective and
allows remote access to my SDR

AlexandreRouma, 2024-01-12:
Running SDR++ in server mode on android is not and will not be supported as it would require bundling an entire terminal emulator to run the headless server or make a UI for a headless mode which is nonsensical.

Using a phone as an SDR server is a bad idea. Running IQ over wifi causes unpredictable latency and leads to buffer underruns which can be heard in the audio.

closed 2024-01-12 by AlexandreRouma

b4545, 2024-01-12:
You're probably right about wireless causing issues.The tcp driver was OK
to me but it just stops streaming when my client disconnects. Thank you for
your time