AlexandreRouma/SDRPlusPlus #1278: SDR++ Nightly Windows 10  Frequency out of place and or completely missing.

gr8lite, 2024-01-14:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
A clear description of the bug.

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

gr8lite, 2024-01-14:
I am using RTL-SDR Blog V4.  
I have set it up in several programs using the appropriate setup instructions.
However after numerous attempts to set it up with SDR++   I have not been able to use the system as it should be.
I get the dongle recognized and select device and settings accordingly.
What happens though is that I can get WFM RADIO stations, but they tune in somewhere in the 50 + MHz area of the tuner.
They come on really well,
I can not find any other frequencies misplaced or correct in any bands.
Using the same settings in multiple software I get all the anticipated results.
I have tried some earlier versions of SDR++ and the Nightly multiple different times.
So far as I can tell I have followed all  the directions, tips, advice, suggestions I could  find.
Please help me determine  the problem  and resolve  it if possible.
I would like to use SDR++ with the RTL-SDR V4 on WIN10 if  it is possible.

AlexandreRouma, 2024-01-14:
1) You're supposed to fill out the issue template not post a second message after it
2) Follow RTL-SDRblog V4 driver replacement on the manufacturer's website: https://www.rtl-sdr.com/v4/

closed 2024-01-14 by AlexandreRouma

gr8lite, 2024-01-14:
OK.
I will try that again.
I may have done it incorrectly.
Thank you for letting me know.

gr8lite, 2024-01-14:
Wow.
I must have  been using the wrong files.
It is working now.
Thanks again for your help.