AlexandreRouma/SDRPlusPlus #1280: Unable to access audio output. audio dlls dont seem to be loading

Soupbones, 2024-01-18:
I am running a Macbook Air with bootcamp running Windows10x64. This laptop has 8Gb of ram, and runs pretty much all my SDR software without any issues. The audio works on all programs I use, as well as VB cable, appart from SDR plus plus.
 I have tried everything but am at a wits end. I lhave two RTL SDR dongles attached, both work fine in other programs with no audio issues.
 I have installed the latest nightly release (v6.02.20 Jan 2024) again after doing a registry clean and a reboot. Yes I'm trying everything and anything.
 So the SDR++ program loads fine. I can set up the RTL SDR and tune it no problem. I can start it, see the spectrum display and water fall. I can tune to say a local FM radio station and I can see everything fine. Gain is up, and I see the signal on the FFT I just cannot hear anything. No audio what so ever and no way to control it.
 The speaker button up the top left is not crossed off and the volume is up. The There i s nothing muted in my windows mixer or in any windows settings.. This computer used the common realtek sound drivers. There is a blue tick next to the radio tab.
In the Sink tab drop down there is no audio there to select an output source. In the modules tab there is no audio source or audio sink options to add in. 
In the SDRplusplus program folder both audio_source.dll and audio_sink.dll are present. They just don't seem to be loading when I run the program.  I can't find a way after extensive searching to force them to load. Only way seems to be to add them in with the modules tab, but they are not present in this tab to add in. I have tried forcing admin to start, have tried compatability modes and tried several older versions.
 Any idea why the audio will not load up to allow me to listen?
 I have zero issues with audio in any other program. and all works fine in programs like  SDR Sharp, HDSDR, SDR Angel and SDR Trunk to name a few.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
A clear description of the bug.

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

AlexandreRouma, 2024-01-19:
Get the startup logs (following the instructions normally meant for a no-start condition in the template you didn't fill out).

Soupbones, 2024-01-19:
Hi.
 As it wasn't a start up issue I didn't do that, but here it is.
[20/01/2024 08:58:39.000] [INFO] SDR++ v1.1.0
[20/01/2024 08:58:39.000] [INFO] Loading config
[20/01/2024 08:58:39.000] [INFO] Using OpenGL 3.0
[20/01/2024 08:58:39.000] [INFO] Loading icons
[20/01/2024 08:58:39.000] [INFO] Loading band plans
[20/01/2024 08:58:39.000] [INFO] Loading band plans color table
[20/01/2024 08:58:39.000] [INFO] Loading modules
[20/01/2024 08:58:39.000] [INFO] Loading C:/SDRplusplus/modules/airspyhf_source.dll
[20/01/2024 08:58:39.000] [INFO] Loading C:/SDRplusplus/modules/airspy_source.dll
[20/01/2024 08:58:39.000] [INFO] Loading C:/SDRplusplus/modules/audio_sink.dll
[20/01/2024 08:58:39.000] [ERROR] Couldn't load C:/SDRplusplus/modules/audio_sink.dll. Error code: 193
[20/01/2024 08:58:39.000] [INFO] Loading C:/SDRplusplus/modules/audio_source.dll
[20/01/2024 08:58:39.000] [ERROR] Couldn't load C:/SDRplusplus/modules/audio_source.dll. Error code: 193
[20/01/2024 08:58:39.000] [INFO] Loading C:/SDRplusplus/modules/bladerf_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/discord_integration.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/file_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/frequency_manager.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/hackrf_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/hermes_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/limesdr_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/m17_decoder.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/meteor_demodulator.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/network_sink.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/perseus_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/plutosdr_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/radio.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/recorder.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/rfspace_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/rigctl_client.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/rigctl_server.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/rtl_sdr_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/rtl_tcp_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/scanner.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/sdrplay_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/sdrpp_server_source.dll
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/soapy_source.dll
[20/01/2024 08:58:40.000] [ERROR] Couldn't load C:/SDRplusplus/modules/soapy_source.dll. Error code: 126
[20/01/2024 08:58:40.000] [INFO] Loading C:/SDRplusplus/modules/spyserver_source.dll
[20/01/2024 08:58:41.000] [INFO] Initializing Airspy Source (airspy_source)
[20/01/2024 08:58:41.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[20/01/2024 08:58:41.000] [INFO] Initializing Audio Sink (audio_sink)
[20/01/2024 08:58:41.000] [ERROR] Module 'audio_sink' doesn't exist
[20/01/2024 08:58:41.000] [INFO] Initializing Audio Source (audio_source)
[20/01/2024 08:58:41.000] [ERROR] Module 'audio_source' doesn't exist
[20/01/2024 08:58:41.000] [INFO] Initializing BladeRF Source (bladerf_source)
[20/01/2024 08:58:41.000] [ERROR] Could not list devices -7
[20/01/2024 08:58:41.000] [INFO] Initializing File Source (file_source)
[20/01/2024 08:58:41.000] [INFO] Initializing Frequency Manager (frequency_manager)
[20/01/2024 08:58:41.000] [INFO] Initializing HackRF Source (hackrf_source)
[20/01/2024 08:58:41.000] [INFO] Initializing Hermes Source (hermes_source)
[20/01/2024 08:58:41.000] [INFO] Initializing LimeSDR Source (limesdr_source)
[20/01/2024 08:58:41.000] [INFO] Initializing Network Sink (network_sink)
[20/01/2024 08:58:41.000] [INFO] Initializing PerseusSDR Source (perseus_source)
perseus: perseus_init()
perseus: EXE: [C:\SDRplusplus\sdrpp.exe]
perseus: path: [C:\SDRplusplus\] name: [sdrpp.exe]
perseus: Found device with VID/PID 05AC:0291 on BUS1 ADDR6
perseus: Found device with VID/PID 05E3:0610 on BUS1 ADDR3
perseus: Found device with VID/PID 1A86:7523 on BUS1 ADDR8
perseus: Found device with VID/PID 046D:C315 on BUS1 ADDR9
perseus: Found device with VID/PID 05E3:0626 on BUS1 ADDR1
perseus: Found device with VID/PID 0BDA:2838 on BUS1 ADDR16
perseus: Found device with VID/PID 1A40:0101 on BUS1 ADDR15
perseus: Found device with VID/PID 0BDA:2838 on BUS1 ADDR17
perseus: Found device with VID/PID 05AC:828F on BUS1 ADDR14
perseus: Found device with VID/PID 0A5C:4500 on BUS1 ADDR5
perseus: Found device with VID/PID 8086:9CB1 on BUS1 ADDR0
perseus: Found device with VID/PID 17EF:6032 on BUS1 ADDR7
[... 41 lines cut ...]
[20/01/2024 08:58:42.000] [INFO] RTLSDRSourceModule 'RTL-SDR Source': Tune: 90790000.000000!
[20/01/2024 08:58:42.000] [INFO] Running post-init for Airspy Source
[20/01/2024 08:58:42.000] [INFO] Running post-init for AirspyHF+ Source
[20/01/2024 08:58:42.000] [INFO] Running post-init for BladeRF Source
[20/01/2024 08:58:42.000] [INFO] Running post-init for File Source
[20/01/2024 08:58:42.000] [INFO] Running post-init for Frequency Manager
[... cut ...]
[20/01/2024 08:58:42.000] [INFO] Running post-init for Radio
[20/01/2024 08:58:42.000] [INFO] Running post-init for Recorder
[20/01/2024 08:58:42.000] [INFO] Running post-init for Rigctl Server
[20/01/2024 08:58:42.000] [INFO] Running post-init for SDR++ Server Source
[20/01/2024 08:58:42.000] [INFO] Running post-init for SDRplay Source
[20/01/2024 08:58:42.000] [INFO] Running post-init for SpyServer Source
[20/01/2024 08:58:42.000] [INFO] Ready.

I can see in this its saying the modules for audio are not there, bu they are in the directory
[image]

AlexandreRouma, 2024-01-19:
> As it wasn't a start up issue I didn't do that, but here it is.

That wasn't the issue, the issue was the template not being filled out which forces me to parse through every sentence to find the information I need instead of it being clearly layed out (See the end of your original post).

In any case, send a zip of that `SDRplusplus` folder, I'll check what's going on.

Soupbones, 2024-01-20:
Here is te hole directory zipped. Its basiclly the files fomr the latest release, though no relase Ive tried worksfor me.
[Uploading SDRplusplus.zip…]()

AlexandreRouma, 2024-01-22:
The link is incorrect, just links to this issue and not a file.

Soupbones, 2024-01-24:
Links to the Issue? You clearly asked me to upload a Zip of the SRDplus Plus folder. Which is what I did.. If I knew exactlt what the issue was, I would send the files.... All I know is the Audio DLL files do not load when the software loads, and I cannot enable them in the sink.

AlexandreRouma, 2024-01-25:
The link in your meesage is not a zip file... it's just a link to this page...
[image]

Soupbones, 2024-01-26:
Says to me Uploading SRDplusplus.zip. Github won't upload the file... ill try again soon. Or ill zip the offending DLLs and try that

Soupbones, 2024-01-26:
Hardware

CPU:Intel(R) Core(TM) i5-5250U CPU @ 1.60GHz   1.60 GHz
RAM:8.00 GB
GPU:Intel UHD Graphics 617
SDR: (Remote or local? If remote, what protocol?) RTL SDR V3
Software 
Operating System: Windows 10 Pro N, 19045.3930
SDR++: v6.02.20 Jan 2024 ( and any previous versions I've tried
Bug Description
Audio.DLLs not loading. No audio working. Unable to load audio sinks in software manually. Audio.DLLs not loading up.

Steps To Reproduce

... See posts above
...
...
Only If SDR++ fails to lauch or the SDR fails to start:
Run SDR++ from a command line window with special parameters:

On Windows, open a terminal and cd to SDR++'s directory and run .\sdrpp.exe -c (if running SDR++ version 1.0.4 or older, use -s instead, though you should probably update SDR++ instead...)
On Linux: Open a terminal and run sdrpp -c
On MacOS: Open a terminal and run /path/to/the/SDR++.app/Contents/MacOS/sdrpp -c
Then, post the entire logs from start to after the issue. DOT NOT truncate to where you think the error is...
Screenshots
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

Additional info
Add any other relevant information.

AlexandreRouma, 2024-01-26:
Well that eplains it, it's an N edition of windows. You need to install the missing media feature pack:
https://support.microsoft.com/en-au/topic/media-feature-pack-list-for-windows-n-editions-c1c6fffa-d052-8338-7a79-a4bb980a700a

closed 2024-01-26 by AlexandreRouma