AlexandreRouma/SDRPlusPlus #1282: Compile under MacOS Catalina 10.15.7 / need help due errors e.g. "filesystem"

aneitzel, 2024-01-18:
SDR++ Compile from source under MacOS Catalina 10.15.7 (Catalina is the latest supported OS-Version on my 27"Imac - thats why i cant use the precompiled artifacts). Downloaded latest GIT [SDR++ Nightly Build (currently 1.1.0)] (https://github.com/AlexandreRouma/SDRPlusPlus/releases/tag/nightly)

I followed readme for compile under MACos, but received a lot of errors and dont know how to react. Most errors are = error: no type named 'path' in namespace 'std::__1::__fs::filesystem

Can someone help me?

################################

**alpha@iMAC build % sudo cmake .. -DUSE_BUNDLE_DEFAULTS=ON -DOPT_OVERRIDE_STD_FILESYSTEM=ON -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_SPYSERVER_SOURCE=OFF -DOPT_BUILD_SDRPLAY_SOURCE=OFF -DOPT_BUILD_HERMES_SOURCE=OFF -DOPT_BUILD_AIRSPY_SOURCE=OFF -DOPT_BUILD_AIRSPYHF_SOURCE=OFF -DOPT_BUILD_SOAPY_SOURCE=ON -DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_AUDIO_SOURCE=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON -DOPT_BUILD_M17_DECODER=ON**
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/build/SDRPlusPlus/build

alpha@iMAC build % sudo make -j
Consolidate compiler generated dependencies of target error_sim_shim
Consolidate compiler generated dependencies of target correct-reed-solomon
Consolidate compiler generated dependencies of target error_sim_sse
Consolidate compiler generated dependencies of target correct-convolutional-sse
Consolidate compiler generated dependencies of target error_sim
Consolidate compiler generated dependencies of target correct-convolutional
Consolidate compiler generated dependencies of target discord-rpc
[  1%] Built target error_sim_shim
[  2%] Built target error_sim
[  5%] Built target error_sim_sse
[  8%] Built target correct-reed-solomon
[ 13%] Built target correct-convolutional-sse
[ 16%] Built target correct-convolutional
[ 16%] Built target correct-sse-h
[ 16%] Built target correct-h
[ 16%] Built target correct_static
[ 18%] Built target correct
[ 22%] Built target discord-rpc
Consolidate compiler generated dependencies of target sdrpp_core
[ 23%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/command_args.cpp.o
[ 23%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/gui/colormaps.cpp.o
[ 23%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/gui/main_window.cpp.o
[ 24%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/core.cpp.o
[ 25%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/gui/theme_manager.cpp.o
[ 26%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/gui/icons.cpp.o
[ 27%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/gui/widgets/folder_select.cpp.o
[ 30%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/gui/style.cpp.o
[ 31%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/server.cpp.o
[ 31%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/gui/widgets/file_select.cpp.o
[ 32%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/module.cpp.o
[ 32%] Building CXX object core/CMakeFiles/sdrpp_core.dir/backends/glfw/backend.cpp.o
In file included from /opt/build/SDRPlusPlus/core/src/command_args.cpp:2:
In file included from /opt/build/SDRPlusPlus/core/std_replacement/filesystem:207:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/fstream:239:41: error: no type named 'path' in namespace 'std::__1::__fs::filesystem'
    basic_filebuf* open(const _VSTD_FS::path& __p, ios_base::openmode __mode) {
                              ~~~~~~~~~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/fstream:1154:47: error: no type named 'path' in namespace 'std::__1::__fs::filesystem'
    explicit basic_ifstream(const filesystem::path& __p, ios_base::openmode __mode = ios_base::in)
                                  ~~~~~~~~~~~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/fstream:1180:33: error: no type named 'path' in namespace 'std::__1::__fs::filesystem'
    void open(const filesystem::path& __p,
                    ~~~~~~~~~~~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/fstream:1368:47: error: no type named 'path' in namespace 'std::__1::__fs::filesystem'
    explicit basic_ofstream(const filesystem::path& __p, ios_base::openmode __mode = ios_base::out)
                                  ~~~~~~~~~~~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/fstream:1395:33: error: no type named 'path' in namespace 'std::__1::__fs::filesystem'
    void open(const filesystem::path& __p, ios_base::openmode __mode = ios_base::out)
                    ~~~~~~~~~~~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/fstream:1582:46: error: no type named 'path' in namespace 'std::__1::__fs::filesystem'
    explicit basic_fstream(const filesystem::path& __p, ios_base::openmode __mode = ios_base::in | ios_base::out)
                                 ~~~~~~~~~~~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/fstream:1610:33: error: no type named 'path' in namespace 'std::__1::__fs::filesystem'
    void open(const filesystem::path& __p, ios_base::openmode __mode = ios_base::in|ios_base::out)
                    ~~~~~~~~~~~~^
7 errors generated.
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/src/command_args.cpp.o] Error 1
[... 343 lines cut ...]
/opt/build/SDRPlusPlus/core/src/signal_path/iq_frontend.h:79:43: note: in instantiation of template class 'dsp::buffer::Reshaper<dsp::complex_t>' requested here
    dsp::buffer::Reshaper<dsp::complex_t> reshape;
                                          ^
/opt/build/SDRPlusPlus/core/src/dsp/buffer/../block.h:65:21: note: overridden virtual function is here
        virtual int run() = 0;
                    ^
7 errors generated.
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/src/gui/style.cpp.o] Error 1
7 errors generated.
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/src/gui/theme_manager.cpp.o] Error 1
3 warnings and 7 errors generated.
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/backends/glfw/backend.cpp.o] Error 1
7 errors generated.
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/src/gui/widgets/folder_select.cpp.o] Error 1
7 errors generated.
1 warning and 7 errors generated.
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/src/gui/widgets/file_select.cpp.o] Error 1
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/src/server.cpp.o] Error 1
1 warning and 7 errors generated.
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/src/gui/main_window.cpp.o] Error 1
1 warning and 7 errors generated.
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/src/core.cpp.o] Error 1
make[1]: *** [core/CMakeFiles/sdrpp_core.dir/all] Error 2
make: *** [all] Error 2
alpha@iMAC build %

AlexandreRouma, 2024-01-18:
Read the instructions mode carefully, the readme says:
> On MacOS versions older than Catalina (10.15)
You are running catalina, not a version older than catalina. You shouldn't use `-DOPT_OVERRIDE_STD_FILESYSTEM=ON`
Delete the build directory and start over.

closed 2024-01-18 by AlexandreRouma

aneitzel, 2024-01-19:
Hi Alexandre, i read those instructions, but instead of not using this option AT ALL i used -DOPT_OVERRIDE_STD_FILESYSTEM=**OFF** ;-)

After deletion of build dir and recompile without DOPT_OVERRIDE_STD_FILESYSTEM IT WORKS !!!!!
Many thanks for fast response

If you have a way to donate to SDR++ in a single donation (not monthly) please let me know!

André