AlexandreRouma/SDRPlusPlus #1299: GPU usage spikes

Tanamichi, 2024-01-25:
Hello there!
Strange behavior if sdrpp launched on Windows and waterfall enabled.
before write this issue i double checked my system for miners and other bloatware, it's clean :) 
I look into task tab and see sdrpp GPU usage high. 

Now i started examine:
1. Waterfall disabled. GPU usage more stable but sometimes spikes GPU usage to 80%
2. Waterfall enabled. GPU usage unstable, spikes come on 80% more periodic.
3. Unpredictable GPU usage. it can more stable without any reasons.
4. Focused or back-grounded sdrpp still invoke GPU usage spikes. 
5. Selected SDR source not affect to spikes. (tried Remote SDR++ server with RTL-SDR and local RSP1 SDR)

it can also affect on other platforms like ARM with weak GPU.

i use these display settings for now and wrote this issue: 
[image]

screenshots below taken in different time.
[image]
[image]
[image]

here is small gif with problem:
[image]

AlexandreRouma, 2024-01-25:
1) Issue template not filled out, If the issue wasn't clear cut I'd have closed it and told you to fill it out properly
2) Your graphics driver is the issue. None of my machines do this, not on windows, not on linux, not on macos and not even on android.

closed 2024-01-25 by AlexandreRouma

Tanamichi, 2024-01-25:
Thanks. Updating to last nvidia driver solved problem :)