AlexandreRouma/SDRPlusPlus #1300: SDRplay not recognized after update 28-1-2024

loewal, 2024-01-28:
**Hardware**
- CPU: Mac Intel
- RAM: 8Gb
- GPU: Intel Core Duo i5
- SDR: SDRplay RSP1

**Software**
- MacOs Monterey 12.7.2
- SDR++: 1.1.0 built 27-1-2024

**Bug Description**
The SDRplay module is not present anymore

**Steps To Reproduce**
1. started earlier version; SDRplay was shown
2. installed named version
3. no more SDRplay selectable

**Additional info**
Add any other relevant information.

AlexandreRouma, 2024-01-28:
Update the SDRplay API to 3.12.1.

closed 2024-01-28 by AlexandreRouma

delboy85341, 2024-02-03:
[AlexandreRouma] commented [last week]

Update the SDRplay API to 3.12.1.

@AlexandreRouma I just went to https://www.sdrplay.com/api/ to download the API.  The version that's there now is 3.14.  I installed it but SDR++ doesn't see my RSP1A.  Thanks for all the work you put into SDR++.  I used to work in IT and had to troubleshoot programs when they didn't work as expected.  I know what a pain in the  backside it can be to track down somehting.

AlexandreRouma, 2024-02-03:
Again, I said 3.12 for a good reason, any version other than 3.12 won't work. You can get 3.12 here
https://www.sdrplay.com/legacy-software/