AlexandreRouma/SDRPlusPlus #1306: SDR++ fails to start (no gui window opens)

lw3eov, 2024-01-29:
**Hardware**
Notebook Sony Vaio VGN-NR130FE
- CPU: Intel Pentium Dual CPU T2330 @ 1.60Ghz codename Merom DC
- Chipset: Intel GL960 (Crestline-GL) + ICH8M version R0130J9
- RAM: 2 GB DDR2-SDRAM 
- GPU:  Intel GM965/GML960 chipset - Integrated graphics 0. Intel GMA X3100. PCI. 384 MB RAM. clock: 400 Mhz
- SDR: local. RTL-SDR original. RTL2832U R820T2 TCX0 + BIAS T + HF., V.3. 

**Software**
- Operating System: Windows 7 ultimate, service pack 1 (build 7601), x64
- SDR++: SDR++ Nightly Build (currently 1.1.0) Nov 22, 2022

Here you can see the output of my sdrpp.exe -c pastebin.com/zLZKqa2F

Error: SDR++ fails to start (no gui window opens)

What I tried but did not help: replacing rtlsdr.dll with the one from here: http://github.com/rtlsdrblog/rtl-sdr-blog/releases/latest/download/Release.zip

AlexandreRouma, 2024-01-29:
I can't tell if this is a joke but anyway thanks for the laugh.
That PC is way too old to run modern software like SDR++
Throw that PC into the garbage where it belongs and get something newer.

closed 2024-01-29 by AlexandreRouma

lw3eov, 2024-01-30:
It is not a joke, I couldn't find much about the system requirements in the documentation. This computer still does a lot of things very well for me and technology is very expensive here so I don't dump any computer. I am going to install an older version of this software.

AlexandreRouma, 2024-01-30:
Older versions of SDR++ won't run either. They require OpenGL 2.1 to 3.0 at the minimum.
Other software like SDR# might start but I doubt they'll run very well. SDR software are CPU intensive.

lw3eov, 2024-01-30:
SDR# works but it is laggy. I've found one that works fine (until now): https://sourceforge.net/projects/u-sdr/
I need to do some more testing, at the moment I only what to receive AFSK as I need to test an APRS TX station we are installing at the local radio club. In a much older computer with Windows XP I have HDSDR (hdsdr.de) but I am getting 9+20 on the S-meter of noise! (which I don't have now on the same location with this other computer + uSDR).

fizr5, 2024-02-08:
Hi. I mentioned this bugreport in https://github.com/AlexandreRouma/SDRPlusPlus/issues/1325 because the developer makes fun of your financial situation and simply the fact, that your hardware is not the newest one.

Here some more helpful information:
There is some graphics functionality that newer computer support your older computer does not fully support. You can try out Linux if you want to run your computer with recent and much more secure operating system. This would also have the benefit, that you could then try out to still run the software on the graphics card in your computer: https://wiki.archlinux.org/title/intel_graphics#OpenGL_2.1_with_i915_driver

timbly5000, 2025-02-13:
What functions provided by OpenGL 3.0 are STRICTLY necessary? I'd argue NONE. Lowering the build requirement to OpenGL 2.1 would help a number of users. Speaking as a DSP professional, I do not buy into the idea that SDR applications are resource hogs - this smacks more of inefficient coding.

AlexandreRouma, 2025-02-13:
> What functions provided by OpenGL 3.0 are STRICTLY necessary? I'd argue NONE.

The *current* code **does** allow for 2.1 with GLSL 1.2 because *such is the requirement of ImGui*. Next version WILL use stuff added in 3.0 for more efficient rendering. Maybe next time don't try to argue about something you don't know.

> I do not buy into the idea that SDR applications are resource hogs - this smacks more of inefficient coding.

Write something faster then, smartass.
And maybe consider the fact that SDR++ outperforms literally every other SDR app in DSP performance...

You can take your pretentious comments and shove them right up where light don't shine. Fuck off.