AlexandreRouma/SDRPlusPlus #1311: Update SDR++ to support RSP-1B

chipperdfw, 2024-02-01:
The new SDRplay RSP-1B is not supported in version 1.1.0.  There appears to be some changes in the API for the RSP-1B.

Hopefully you can update to support in the near future.

https://www.sdrplay.com/rsp1b/

AlexandreRouma, 2024-02-01:
I don't have one so can't test/validate anything. Will be a while before this can be added.

loewal, 2024-02-02:
This is my last working version with a RSP1a
[image]

chipperdfw, 2024-02-02:
RSP1B with SDRplay API v3.14 on Windows 10.
[image]

Would you like me to test with RSP-1A and API v3.14?  I think I used v3.12 previously.

Dustify, 2024-02-09:
'appreciate you don't take PRs, link below essentially clones the RSP1A functionality as I don't believe there's much difference. I'm willing to be proven wrong.

https://github.com/Dustify/SDRPlusPlus/blob/rsp1b/source_modules/sdrplay_source/src/main.cpp

It appears to be behaving itself for me with limited testing.

JordanSavoie, 2024-02-09:
Tried @Dustify's suggestion, but he didn't have `OPT_BUILD_SDRPLAY_SOURCE` set to `ON` in CMakeLists.txt. I did that, but it doesn't recognize my RSP1B. When building the MacOS bundle it says at one point

`Installing ./build/source_modules/sdrplay_source/sdrplay_source.dylib
==NOT== Installing libsdrplay_api.so.3.14`

but it seems to also does the same if I go to version 3.12 with the RSP1B stuff removed. I don't have an RTL or something on me to see if my build is even working correctly, but I can go try tomorrow.

[image]

Dustify, 2024-02-10:
@JordanSavoie rather than change the source, you can specify options when you run cmake, e.g.

`cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=true`

Can you check under "Module Manager" to see if you have an instance of "sdrplay_source". If not, does it appear in the list at the bottom so you can add it?

[image]

JordanSavoie, 2024-02-10:
It doesn't show up in my module manager nor the list, neither in my build or in the nightly release.

[image]

JordanSavoie, 2024-02-12:
@Dustify Yeah I have it installed. I'm on M1 and don't really know what I'm doing anyway, so it's not too surprising that I wasn't able to get it working :p. Alexandre seems to be doing some work with an ARM build, so maybe he'll find something wrong

ebirn, 2024-02-16:
@Dustify I did exactly the same modifications, and have it running on MacOS 14.2.1 (Intel)
i.e. see https://github.com/AlexandreRouma/SDRPlusPlus/compare/master...ebirn:SDRPlusPlus:add_sdrplay_rsp1b

gerner, 2024-02-20:
I also made similar changes (i.e. treat rsp1b as if it behaves exactly the same as rsp1a) and it appears to work for me on Ubuntu 20.04 AMD64.

nmaster2042, 2024-03-22:
If it seems to work like RSP1A, I'm wondering if gains on some bands are the same or not.
It's generally something that is different on each RSP model.
And for RSP1B they oversized better performance on bands like:
- below 1MHz  (i.e for LF and below)
- 50-60 MHz region
- noticeable noise improvements in the 3.5-5.5MHz and 250-320MHz
Do you have experienced gain settings on these bands ?

thebrunx, 2024-03-28:
What a bummer. I have the exact same problem with my new RSP1B on Mac M1, SDRplay is not in the list. so can't get it to work. funny enough it does work on SDRangel and SDRconnect without a problem. still no fix for SDR++? I love ++ and would like to keep using it,  please help.

gerner, 2024-03-28:
@thebrunx sdrplay support isn't turned on in the release build. you need to build from source and include the build option for sdrplay. You can check out the readme for details, but basically when you run cmake, you want a command line something like `cmake -DOPT_BUILD_SDRPLAY_SOURCE=1`

Once you do that, it still won't recognize your RSP1B. It'll still let you use it as an unknown SDRPlay device. That actually sort of works. But you might need to fiddle with the options to find a set that works. This is very much not supported, and anything that happens, good or bad is up to you. Good luck.

thebrunx, 2024-03-28:
Unfortunately I'm not much of a programmer and have never "built" before, Is it possible to have video/YouTube instructions on how to do this?

gerner, 2024-03-28:
@thebrunx try the discord server, maybe someone there can help you: https://discord.gg/aFgWjyD

jimcj1301, 2024-07-21:
Hello, 

I'm new to the RTL-SRD community. 

Apologize if I missed it somewhere but is there a software package for Apple Silicon yet? I have a M1 Pro Max
 Thank you!

KubaPro010, 2024-07-31:
> Would this SDRplay clone work the mentioned patch? https://www.aliexpress.com/item/1005007134319517.html Chip: MSI 001+MSI 2500 chip Model: FSP1A Pro

That doesn't need a patch as its detected as a RSP1 (i have a very similiar model, RSP1C which is detected as a RSP1)

> Hello,
> 
> I'm new to the RTL-SDR community.
> 
> Apologize if I missed it somewhere but is there a software package for Apple Silicon yet? I have a M1 Pro Max Thank you!
Off-topic, also https://github.com/AlexandreRouma/SDRPlusPlus/releases/tag/nightly ([sdrpp_macos_arm.zip](https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/nightly/sdrpp_macos_arm.zip)) (and not even rtl-sdr but sdr community, sdr++ isnt for rtlsdr only)

AlexandreRouma, 2025-02-13:
Support was added a few months ago, forgot to close.

closed 2025-02-13 by AlexandreRouma

lolo83160, 2025-02-28:
Bonjour à tous  j’ai actuellement investi dans un SSRplay  rspdx-r2  l’original que je conseille fortement rien à voir avec les clone,  savez-vous comment le faire reconnaître sur SDRsharp pour le moment je n’ai pas de solution,  et par la suite il serait sûrement bien de pouvoir decoer tétra et d’autre sources via SDRuno et connect, si vous avez ûe idée je suis preneur à bientôt merci d’avance à tous les développeurs vous êtes au top vraiment 😉👍