AlexandreRouma/SDRPlusPlus #1313: Android and hackrf crash after play

hudy000, 2024-02-01:
**Hardware**
Samsung galaxy s22 ultra (android 14)
Samsung galaxy Tab A8 (android 13)
HackRF one

**Software**
- sdr++ for android 

**Bug Description**
If i run app, select hack rf, refresh for driver visible and click play, app crashes every time.

**Steps To Reproduce**
1. Try diferent phone and tablet
2. ...testing other drivers

https://github.com/AlexandreRouma/SDRPlusPlus/assets/156708817/27b26ba8-76db-43b8-8e46-8ace496416bd

AlexandreRouma, 2024-02-01:
can't replicate on my note 9

hudy000, 2024-02-01:
> can't replicate on my note 9

Try yt

https://youtu.be/6FZdZ9aQL7g?si=lhXVWhwv0rBiEmLa

hudy000, 2024-02-09:
How can i help solved this problem ?
Sdr touch working without crash

GrieverGraver, 2024-02-16:
Got same issue at huawei matepad wgr-w09, harmony 2.0.0

buccinator, 2024-02-18:
same issue on samsung s23 ultra

drafthost1, 2024-03-19:
same issue on pixel 7 pro

pizvizdik, 2024-04-04:
same problem on motorola razr 40, even a "hack" with removing the soapy module doesn't help - keeps crashing

izeroo, 2024-04-21:
Can confirm on Redmi K60 running Android 14 with Hackrf One
Stack trace:
```
Redmi/mondrian/mondrian:14/UKQ1.230804.001/V816.0.6.0.UMNCNXM:user/release-keys

      #00 pc 00000000000fe098  /apex/com.android.runtime/lib64/bionic/libc.so (pthread_mutex_lock+16) (BuildId: 1e3ca19bcae05c01b019c85f3f422e56)
      #01 pc 000000000000dea8  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libusb1.0.so (offset 0x478000) (BuildId: 0e0d0e4de78c3d0ac3b0ed8a0da0ba7f5e0d4333)
      #02 pc 00000000000118fc  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libusb1.0.so (offset 0x478000) (BuildId: 0e0d0e4de78c3d0ac3b0ed8a0da0ba7f5e0d4333)
      #03 pc 000000000000a324  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libusb1.0.so (offset 0x478000) (libusb_wrap_sys_device+136) (BuildId: 0e0d0e4de78c3d0ac3b0ed8a0da0ba7f5e0d4333)
      #04 pc 0000000000003cf8  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libhackrf.so (offset 0xd30000) (hackrf_open_by_fd+96) (BuildId: 37fe2cc12078c129f368097375fc990f7f01c360)
      #05 pc 000000000000c618  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!hackrf_source.so (offset 0xfc2000) (HackRFSourceModule::start(void*)+112) (BuildId: cbae4b5507877d4115294b546307e8b73eff8884)
      #06 pc 00000000002169fc  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libsdrpp_core.so (offset 0x11e8000) (SourceManager::start()+24) (BuildId: bec9e65e8a9a1d405ae99a95c28c4a8326184491)
      #07 pc 000000000014cf5c  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libsdrpp_core.so (offset 0x11e8000) (MainWindow::draw()+2080) (BuildId: bec9e65e8a9a1d405ae99a95c28c4a8326184491)
      #08 pc 0000000000226524  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libsdrpp_core.so (offset 0x11e8000) (backend::renderLoop()+320) (BuildId: bec9e65e8a9a1d405ae99a95c28c4a8326184491)
      #09 pc 000000000012b9d0  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libsdrpp_core.so (offset 0x11e8000) (sdrpp_main(int, char**)+23368) (BuildId: bec9e65e8a9a1d405ae99a95c28c4a8326184491)
      #10 pc 000000000022717c  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libsdrpp_core.so (offset 0x11e8000) (android_main+304) (BuildId: bec9e65e8a9a1d405ae99a95c28c4a8326184491)
      #11 pc 0000000000228fb4  /data/app/~~s4nlY-WUkGJ3JgbLyRvc0g==/org.sdrpp.sdrpp-GDNNs316mUQE-d5j2txJEw==/base.apk!libsdrpp_core.so (offset 0x11e8000) (BuildId: bec9e65e8a9a1d405ae99a95c28c4a8326184491)
      #12 pc 00000000000fd134  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+208) (BuildId: 1e3ca19bcae05c01b019c85f3f422e56)
      #13 pc 0000000000096ae4  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+68) (BuildId: 1e3ca19bcae05c01b019c85f3f422e56)

```

izeroo, 2024-04-21:
> Can confirm on Redmi K60 running Android 14 with Hackrf One Stack trace:
>
> ```
> Redmi/mondrian/mondrian:14/UKQ1.230804.001/V816.0.6.0.UMNCNXM:user/release-keys
>
>       #00 pc 00000000000fe098 [...]
April 16, 2024 Build

pinjuf, 2024-07-25:
Can also confirm on an up-to-date Samsung A55 5G (Android 14), running the current (nightly) build (1.2.0). Recognizes my HackRF (Clone r10 + Portapack H2), crashes when "Play" is pressed. Works with my RTL-SDR v4 though.

Cricre, 2024-08-10:
Hi 
I have exactly the same problem : even if connecting Hack RF with no antenna attached , it crashes every time on my pixel 6a, android 14, apk version of 07.08.2024. 

What can we provide you as info to help detect the bug ?

Best regards

hudy000, 2024-08-31:
It seems like they don't care about us, so much time has passed, and it still doesn't work.

duckyondiscord, 2024-09-17:
bumpty bump
same on my pixel 7 running calyxOS

Zarale00, 2024-10-08:
anyone having this issue, it seems that app have weird issue with sepolicy, so putting your android device from enforcing to permissive worked for me, should work for you too but root access is needed

duckyondiscord, 2024-10-14:
> anyone having this issue, it seems that app have weird issue with sepolicy, so putting your android device from enforcing to permissive worked for me, should work for you too but root access is needed

this worked tysm!

hudy000, 2024-10-14:
Hi, can you tell something more ?Wysłano z mojego urządzenia Galaxy
-------- Oryginalna wiadomość --------Od: ducky ***@***.***> Data: 14.10.2024  15:09  (GMT+01:00) Do: AlexandreRouma/SDRPlusPlus ***@***.***> DW: hudy000 ***@***.***>, Author ***@***.***> Temat: Re: [AlexandreRouma/SDRPlusPlus] Android and hackrf crash after play (Issue #1313) 

anyone having this issue, it seems that app have weird issue with sepolicy, so putting your android device from enforcing to permissive worked for me, should work for you too but root access is needed

this worked tysm!

duckyondiscord, 2024-10-14:
if you're rooted with magisk, the easiest way is to just run `setenforce 0` as root, then write it to a script under /data/adb/post-fs-data.d/whatever.sh and it should disable selinux every time you boot up

hudy000, 2024-10-14:
Oh, ok. Unfortunately, I can't root my device because I have Knox state active.
-------- Oryginalna wiadomość --------Od: ducky ***@***.***> Data: 14.10.2024  19:22  (GMT+01:00) Do: AlexandreRouma/SDRPlusPlus ***@***.***> DW: hudy000 ***@***.***>, Mention ***@***.***> Temat: Re: [AlexandreRouma/SDRPlusPlus] Android and hackrf crash after play (Issue #1313) 
@hudy000 if you're rooted with magisk, the easiest way is to just run setenforce 0 as root, then write it to a script under /data/adb/post-fs-data.d/whatever.sh and it should disable selinux every time you boot up

AlexandreRouma, 2026-04-16:
@bubnikv Can you not spam your fork over 10 different issues? If you continue this crap you're getting banned from the repo, I'm tired of the spam.

demirtronics, 2026-10-09:
Root cause: android-sdr-kit builds libhackrf with the old option name `-DDISABLE_USB_ENUMERATION=ON`, which was renamed to `DISABLE_USB_DEVICE_DISCOVERY` in the hackrf fork. So `LIBUSB_OPTION_NO_DEVICE_DISCOVERY` is never set, `libusb_init()` fails under SELinux enforcing (hence `setenforce 0` helps), and `hackrf_open_by_fd()` crashes on a NULL libusb context.

One-line fix: AlexandreRouma/android-sdr-kit#3. Verified on Redmi Note 8 / Android 11 with HackRF One by swapping a rebuilt `libhackrf.so` into the current nightly APK.