AlexandreRouma/SDRPlusPlus #1317: Adalm pluto via ethernet not found

Denisfk1985, 2024-02-02:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: LibreSDR (Adalm pluto) via Ethernet

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).
Windows 10 x64
SDR++ v1.1.0 build 02 feb 2024

**Bug Description**

Adalm pluto via ethernet(LibreSDR) not found in current nightly versiot at windows.
But in old version sdrpp i can it is possible to specify the IP address of the pluto, but it not avaible in current version sdrpp.

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

AlexandreRouma, 2024-02-02:
Show logs

F5OEO, 2024-02-05:
Libresdr is a derivative of plutosdr. Even it is compatible with PlutoSDR, its signature is different : for example, model is ""LibreSDR Rev.5 (Z7020/AD9363)"" and not "Analog Devices PlutoSDR Rev.B (Z7010-AD9364)" . Same behaviour with Antsdr. 

Propose solution is to be less strict and just show a warning here : 

https://github.com/AlexandreRouma/SDRPlusPlus/blob/f66f2c25e1d35f675db86905b1166c25fb929f42/source_modules/plutosdr_source/src/main.cpp#L133

AlexandreRouma, 2024-02-05:
@F5OEO It's why I'm asking for logs to check if it was being rejecte., But no, you can't just put a warning there. PlutoSDR & derivatives aren't the only thing that use IIO, here is a concrete example of the wrong device being shown when this check was not in place:
[image]

F5OEO, 2024-02-05:
Here are antsdr and plutosdr on network and usb.
[image]

Maybe string filter on "Analog Devices" instead of "PlutoSDR"...or add each variant string (antsdr,libresdr,plutoplus) ?

AlexandreRouma, 2024-02-05:
I've got an AntSDR on the way, should get it thursday, will experiment with things like that and see what works best.

kiklhorn, 2024-02-10:
PlutoSDR on LAN address 192.168.199.67 
Working on version 1.0.4
Not discovered on version 1.1.0 

Is there any way how to add IP manually?
[image]

Logs:
`C:\Users\marti\Downloads\sdrpp_windows_x64-1.0.4>sdrpp.exe -s
[2024-02-10 10:29:36.640] [info] SDR++ v1.0.4
[2024-02-10 10:29:36.643] [info] Loading config
[2024-02-10 10:29:37.054] [info] Using OpenGL 3.0
[2024-02-10 10:29:37.187] [info] Loading icons
[2024-02-10 10:29:37.274] [info] Loading band plans
[2024-02-10 10:29:37.282] [info] Loading band plans color table
[2024-02-10 10:29:37.296] [info] Loading modules
[2024-02-10 10:29:37.297] [info] Loading ./modules/airspyhf_source.dll
[2024-02-10 10:29:37.369] [info] Loading ./modules/airspy_source.dll
[2024-02-10 10:29:37.416] [info] Loading ./modules/audio_sink.dll
[2024-02-10 10:29:37.471] [info] Loading ./modules/bladerf_source.dll
[2024-02-10 10:29:37.517] [info] Loading ./modules/discord_integration.dll
[2024-02-10 10:29:37.544] [info] Loading ./modules/file_source.dll
[2024-02-10 10:29:37.573] [info] Loading ./modules/frequency_manager.dll
[2024-02-10 10:29:37.605] [info] Loading ./modules/hackrf_source.dll
[2024-02-10 10:29:37.650] [info] Loading ./modules/limesdr_source.dll
[2024-02-10 10:29:37.708] [info] Loading ./modules/m17_decoder.dll
[2024-02-10 10:29:37.753] [info] Loading ./modules/meteor_demodulator.dll
[2024-02-10 10:29:37.785] [info] Loading ./modules/network_sink.dll
[2024-02-10 10:29:37.814] [info] Loading ./modules/plutosdr_source.dll
[2024-02-10 10:29:37.883] [info] Loading ./modules/radio.dll
[2024-02-10 10:29:37.914] [info] Loading ./modules/recorder.dll
[2024-02-10 10:29:37.951] [info] Loading ./modules/rigctl_server.dll
[2024-02-10 10:29:37.980] [info] Loading ./modules/rtl_sdr_source.dll
[2024-02-10 10:29:38.035] [info] Loading ./modules/rtl_tcp_source.dll
[2024-02-10 10:29:38.066] [info] Loading ./modules/sdrplay_source.dll
[2024-02-10 10:29:38.100] [info] Loading ./modules/soapy_source.dll
[2024-02-10 10:29:38.132] [error] Couldn't load ./modules/soapy_source.dll. Error code: 126
[2024-02-10 10:29:38.133] [info] Loading ./modules/spyserver_source.dll
[2024-02-10 10:29:38.163] [info] Initializing Airspy Source (airspy_source)
[2024-02-10 10:29:40.415] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2024-02-10 10:29:41.751] [info] Initializing Audio Sink (audio_sink)
[2024-02-10 10:29:41.753] [info] Initializing BladeRF Source (bladerf_source)
[2024-02-10 10:29:41.829] [error] Could not list devices -7
[2024-02-10 10:29:41.830] [info] Initializing File Source (file_source)
[2024-02-10 10:29:41.833] [info] Initializing Frequency Manager (frequency_manager)
[2024-02-10 10:29:41.835] [info] Initializing HackRF Source (hackrf_source)
[2024-02-10 10:29:41.908] [info] Initializing LimeSDR Source (limesdr_source)
[2024-02-10 10:29:41.911] [info] Initializing Network Sink (network_sink)
[2024-02-10 10:29:41.913] [info] Initializing PlutoSDR Source (plutosdr_source)
[2024-02-10 10:29:41.915] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2024-02-10 10:29:44.425] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2024-02-10 10:29:44.428] [info] Initializing Radio (radio)
[2024-02-10 10:29:45.623] [info] RtAudio stream open
[2024-02-10 10:29:45.624] [info] Initializing Recorder (recorder)
[2024-02-10 10:29:45.628] [info] Initializing Rigctl Server (rigctl_server)
[2024-02-10 10:29:45.630] [info] Initializing SDRplay Source (sdrplay_source)
[2024-02-10 10:29:45.632] [error] Could not intiatialized the SDRplay API. Make sure that the service is running.
[2024-02-10 10:29:45.632] [info] Initializing SoapySDR Source (soapy_source)
[2024-02-10 10:29:45.634] [error] Module 'soapy_source' doesn't exist
[2024-02-10 10:29:45.638] [info] Initializing SpyServer Source (spyserver_source)
[2024-02-10 10:29:45.641] [info] Loading color maps
[2024-02-10 10:29:45.643] [info] Loading ./res/colormaps/classic.json
[2024-02-10 10:29:45.646] [info] Loading ./res/colormaps/classic_green.json
[2024-02-10 10:29:45.648] [info] Loading ./res/colormaps/electric.json
[2024-02-10 10:29:45.653] [info] Loading ./res/colormaps/gqrx.json
[2024-02-10 10:29:45.655] [info] Loading ./res/colormaps/greyscale.json
[2024-02-10 10:29:45.656] [info] Loading ./res/colormaps/inferno.json
[2024-02-10 10:29:45.658] [info] Loading ./res/colormaps/magma.json
[2024-02-10 10:29:45.662] [info] Loading ./res/colormaps/plasma.json
[2024-02-10 10:29:45.665] [info] Loading ./res/colormaps/turbo.json
[... 162 lines cut ...]
[10/02/2024 10:32:43.000] [INFO] Running post-init for AirspyHF+ Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for Audio Sink
[10/02/2024 10:32:43.000] [INFO] Running post-init for Audio Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for BladeRF Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for File Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for Frequency Manager
[10/02/2024 10:32:43.000] [INFO] Running post-init for HackRF Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for Hermes Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for LimeSDR Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for Network Sink
[10/02/2024 10:32:43.000] [INFO] Running post-init for PerseusSDR Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for PlutoSDR
[10/02/2024 10:32:43.000] [INFO] Running post-init for RFspace Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for RTL-SDR Source
[... cut ...]
[10/02/2024 10:32:43.000] [INFO] Running post-init for SDR++ Server Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for SDRplay Source
[10/02/2024 10:32:43.000] [INFO] Running post-init for SpyServer Source
[10/02/2024 10:32:43.000] [INFO] Ready.
[10/02/2024 10:32:57.000] [INFO] PlutoSDRSourceModule 'PlutoSDR': Tune: 100000000.000000!
[10/02/2024 10:33:02.000] [INFO] PlutoSDRSourceModule 'PlutoSDR': Tune: 103525000.000000!
[10/02/2024 10:33:06.000] [INFO] PlutoSDRSourceModule 'PlutoSDR': Tune: 107025000.000000!`

AlexandreRouma, 2024-02-10:
Ok so it looks like iio itself isn't even finding the device, You can check that by running `iio_info.exe -s`

> Is there any way how to add IP manually?

There is not, the point of switching to auto discovery is to fix the hack that was to force users to manually specify an IP and have proper support for per-device settings.

AlexandreRouma, 2024-02-13:
Alright it seems the issue is the version of IIO shipped with the windows build is missing network autodetection for some reason. I guess this will only be fixed once I move to my own build of it... Stay tuned I guess

Sprutx, 2024-03-28:
Hi, I'm using LibreSDR (aka PlutoSDR) and I'm seeing the same error - SDR++ doesn't find the device

Debian 12
SDR++ Nightly Build (currently 1.1.0) 03/26/2024

log:
```
[28/03/2024 14:06:06.000] [INFO] RtAudio stream open
[28/03/2024 14:06:06.000] [INFO] AirspySourceModule 'Airspy Source': Tune: 100000000.000000!
[28/03/2024 14:06:06.000] [INFO] Running post-init for Airspy Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for AirspyHF+ Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for Audio Sink
[28/03/2024 14:06:06.000] [INFO] Running post-init for Audio Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for BladeRF Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for File Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for Frequency Manager
[28/03/2024 14:06:06.000] [INFO] Running post-init for HackRF Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for Hermes Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for Network Sink
[28/03/2024 14:06:06.000] [INFO] Running post-init for PlutoSDR Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for RFspace Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for RTL-TCP Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for Radio
[28/03/2024 14:06:06.000] [INFO] Running post-init for Recorder
[28/03/2024 14:06:06.000] [INFO] Running post-init for Rigctl Server
[28/03/2024 14:06:06.000] [INFO] Running post-init for SDR++ Server Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for SoapySDR Source
[28/03/2024 14:06:06.000] [INFO] Running post-init for SpyServer Source
[28/03/2024 14:06:06.000] [INFO] Ready.
[28/03/2024 14:07:36.000] [INFO] AirspySourceModule 'Airspy Source': Menu Deselect!
[28/03/2024 14:07:36.000] [INFO] New DSP samplerate: 4000000.000000 (source samplerate is 4000000.000000)
[28/03/2024 14:07:36.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Menu Select!
[28/03/2024 14:07:45.000] [WARN] Ignored IIO device: [ip:libre.local] 192.168.1.10 (LibreSDR Rev.5 (Z7020-AD9361)), serial=
[28/03/2024 14:10:56.000] [INFO] Exiting successfully
```

AlexandreRouma, 2024-04-08:
First, version 1.0.4 doesn't "find" anything because it doesn't search for devices in the first place, your pluto just has the default IP.

Secondly, the issue is completely different to this one. Yours is that SDR++ expects "PlutoSDR" devices, not "LibreSDR". I'm not sure yet how to best handle this and I suggest you open a separate bug report.

EeeLo, 2024-07-17:
Same question about it,Debian 8.x
[image]

log:
[16/07/2024 20:39:24.000] [INFO] SDR++ v1.2.0
[16/07/2024 20:39:24.000] [INFO] Loading config
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen
[16/07/2024 20:39:24.000] [INFO] Using OpenGL 3.0
[16/07/2024 20:39:24.000] [INFO] Loading icons
[16/07/2024 20:39:24.000] [INFO] Loading band plans
[16/07/2024 20:39:24.000] [INFO] Loading band plans color table
[16/07/2024 20:39:24.000] [INFO] Loading modules
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/pager_decoder.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/iq_exporter.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[16/07/2024 20:39:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[16/07/2024 20:39:24.000] [INFO] Initializing Airspy Source (airspy_source)
[16/07/2024 20:39:24.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[16/07/2024 20:39:24.000] [ERROR] Module 'airspyhf_source' doesn't exist
[16/07/2024 20:39:24.000] [INFO] Initializing Audio Sink (audio_sink)
[16/07/2024 20:39:24.000] [INFO] Initializing Audio Source (audio_source)
[16/07/2024 20:39:24.000] [INFO] Initializing BladeRF Source (bladerf_source)
[16/07/2024 20:39:24.000] [ERROR] Module 'bladerf_source' doesn't exist
[16/07/2024 20:39:24.000] [INFO] Initializing File Source (file_source)
[16/07/2024 20:39:24.000] [INFO] Initializing Frequency Manager (frequency_manager)
[16/07/2024 20:39:24.000] [INFO] Initializing HackRF Source (hackrf_source)
[16/07/2024 20:39:24.000] [INFO] Initializing Hermes Source (hermes_source)
[16/07/2024 20:39:24.000] [INFO] Initializing LimeSDR Source (limesdr_source)
[16/07/2024 20:39:24.000] [ERROR] Module 'limesdr_source' doesn't exist
[16/07/2024 20:39:24.000] [INFO] Initializing Network Sink (network_sink)
[16/07/2024 20:39:24.000] [INFO] Initializing PerseusSDR Source (perseus_source)
[16/07/2024 20:39:24.000] [ERROR] Module 'perseus_source' doesn't exist
[16/07/2024 20:39:24.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
ERROR: Unable to create Avahi DNS-SD client :Daemon not running
[16/07/2024 20:39:25.000] [ERROR] Failed to enumerate contexts
[16/07/2024 20:39:25.000] [INFO] Initializing RFNM Source (rfnm_source)
[16/07/2024 20:39:25.000] [ERROR] Module 'rfnm_source' doesn't exist
[16/07/2024 20:39:25.000] [INFO] Initializing RFspace Source (rfspace_source)
[16/07/2024 20:39:25.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[16/07/2024 20:39:25.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[16/07/2024 20:39:25.000] [INFO] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 32, interp: 5000, decim: 7813, inacc: 0.000000%, taps: 593750
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[16/07/2024 20:39:25.000] [WARN] Demod switch took 59745 us
[16/07/2024 20:39:25.000] [INFO] RtAudio stream open
[... 29 lines cut ...]
[16/07/2024 20:39:26.000] [INFO] Running post-init for Audio Source
[16/07/2024 20:39:26.000] [INFO] Running post-init for File Source
[16/07/2024 20:39:26.000] [INFO] Running post-init for Frequency Manager
[16/07/2024 20:39:26.000] [INFO] Running post-init for HackRF Source
[16/07/2024 20:39:26.000] [INFO] Running post-init for Hermes Source
[16/07/2024 20:39:26.000] [INFO] Running post-init for Network Sink
[16/07/2024 20:39:26.000] [INFO] Running post-init for PlutoSDR Source
[16/07/2024 20:39:26.000] [INFO] Running post-init for RFspace Source
[16/07/2024 20:39:26.000] [INFO] Running post-init for RTL-SDR Source
[... cut ...]
[16/07/2024 20:39:30.000] [INFO] New DSP samplerate: 4000000.000000 (source samplerate is 4000000.000000)
[Resamp] predec: 8, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[16/07/2024 20:39:30.000] [INFO] New DSP samplerate: 2500000.000000 (source samplerate is 2500000.000000)
ERROR: Unable to create Avahi DNS-SD client :Daemon not running
[16/07/2024 20:39:32.000] [ERROR] Failed to enumerate contexts
[Resamp] predec: 8, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[16/07/2024 20:39:32.000] [INFO] New DSP samplerate: 2500000.000000 (source samplerate is 2500000.000000)

Karsonito, 2024-08-25:
Temp workaround found:
Replace libiio.dll inside sdrpp_windows_x64 with any build from https://github.com/analogdevicesinc/libiio/releases/tag/v0.25
New versions of libiio contains autodetection

> Alright it seems the issue is the version of IIO shipped with the windows build is missing network autodetection for some reason. I guess this will only be fixed once I move to my own build of it... Stay tuned I guess
Hello @AlexandreRouma!
Any news on fixing inside new windows builds?

0x0V3RF10W, 2024-08-25:
Temp workaround found:
Replace libiio.dll inside sdrpp_windows_x64 with any build from https://github.com/analogdevicesinc/libiio/releases/tag/v0.25
New versions of libiio contains autodetection

>Alright it seems the issue is the version of IIO shipped with the windows build is missing network autodetection for some reason. I guess this will only be fixed once I move to my own build of it... Stay tuned I guess

Hello @AlexandreRouma !
Any news on fixing inside new windows builds?

Yura1993, 2024-09-07:
> Temp workaround found: Replace libiio.dll inside sdrpp_windows_x64 with any build from https://github.com/analogdevicesinc/libiio/releases/tag/v0.25 New versions of libiio contains autodetection [...]

Can you please explain how to replace this libiio.dll inside sdrpp_windows_x64? i try to replace it dircetly - after that SDR++ try to make boot and stuck on "PlutoSDR start". I use last SDRpp build and libiio v0.26. Try to make copy this libiio.dll from all 3 folders - nothing result, all the same. 
 
 And by the way - original libiio.dll that preinstalled on sdrpp is have 1615kB size, and all libiio.dll from analog device repository have or 108kB or 626kB, that mean something wrong. Maby i need to take it from another place?

 Please make some explanation bcs  @AlexandreRouma doesn't solve this issue

0x0V3RF10W, 2024-09-07:
> Can you please explain how to replace this libiio.dll inside sdrpp_windows_x64?

Just replace libiio.dll and that's it - any version is compatiable (MinGW, VS)

> i try to replace it dircetly - after that SDR++ try to make boot and stuck on "PlutoSDR start".

I've only seen such stuck once as far as I can remember and just restarted sdr++. Now it is very stable

- Check your firewall settings
- Run `iio_info -S` and copy here

Yura1993, 2024-09-08:
> Just replace libiio.dll and that's it - any version is compatiable (MinGW, VS)
>
>
> I've only seen such stuck once as far as I can remember and just restarted sdr++. Now it is very stable
>
> * Check your [...]

 So as i say previously - i try to replace it, and its doesn't work, stuck on boot SDR++ etc.
After that i try to take another libiio.dll, and take it from SDRConsole folder, after this replace - its start to work fine (with a little interrupt from beginning, but actually works! and find usb and ethernet devices)

 Version of libiio.dll that i take from SDRConsole was 0.24, all version after - doesn't work. Why? i don't know 

 Firewall setting allow connection

 P.S. Do you hear something about support of plutosdr(and his clones) of transmitting on SDR++? I need at least simple plugin with basic modulation support, its will be great

0x0V3RF10W, 2024-09-08:
> I use last SDRpp build

@Yura1993 which version exactly?

> Version of libiio.dll that i take from SDRConsole was 0.24, all version after - doesn't work. Why? i don't know

I checked libiio.dll from sdr console 3.3 without any problems
Windows 11

ashjmckenzie, 2024-11-17:
I can confirm this on latest windows build,
I have LibreSDR works perfect on 1.0.4, I could not get the workaround to function, any time I replace the DLL the plutoSDR option disappears from the SDR++ device list, 

When I run Libio it autodetects the SDR, its on a manually set IP in my subnet

ashjmckenzie, 2024-11-17:
I figured out the work around..... here is what I did:

Download https://github.com/analogdevicesinc/libiio/releases/download/v0.26/Windows.zip

From: Windows.zip\Windows-VS-2022-x64
Copy the following to the lastest build:
- libserialport-0.dll
- libiio-sharp.dll
- libiio.dll (this will overight)
- libxml2.dll

Then SDR++ will autodetect PlutoIP Devices (including LibreSDR)

Yura1993, 2025-01-08:
> I figured out the work around..... here is what I did:
>
> Download https://github.com/analogdevicesinc/libiio/releases/download/v0.26/Windows.zip
>
> From: Windows.zip\Windows-VS-2022-x64 Copy the [...]

Unfortunately this dosent work with PuzhiSDR P203mini, and without manual option of adding device by ip address
On SDR# this works fine

VasylSamoilov, 2025-01-08:
Can confirm, not detected and changing dll's does nothing. Latest nightly.

povnz, 2025-03-24:
Ok so the issue is with IPv4 and IPv6 being both avaible, AlphaFox2 has proposed a fix: https://github.com/AlexandreRouma/SDRPlusPlus/issues/1594

But before this gets merged, 

Download the Nightly
Patch the DLLs as listed above
Ensure IPv6 is disable on the NIC being used

SDR++ will work!

AlexandreRouma, 2026-10-23:
Should be fixed in the latest windows nightly.

closed 2026-10-23 by AlexandreRouma