AlexandreRouma/SDRPlusPlus #1322: SDR++ nightly (1.1.0-1220) failed to initialize OpenGL with GLSL and dont start

fizr5, 2024-02-07:
I hope its enough to quickly show the logfiles because the combination dont make sense:

OS: Lubuntu 22.04 running X11 (not wayland) on a computer with 8GB RAM (for sure no RAM issue)
Installed SDR++ package from github (sdrpp_ubuntu_jammy_amd64.deb (1.1.0-1220) )

glxinfo | grep "OpenGL version"
OpenGL version string: **2.1** Mesa 23.0.4-0ubuntu1~22.04.1

SDR++ logfile:
[INFO] SDR++ v1.1.0
[INFO] Loading config
[WARN] ConfigManager locked, waiting...
[ERROR] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[INFO] OpenGL 3.0 was not supported
[ERROR] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[INFO] OpenGL 3.1 ES was not supported
[INFO] Using OpenGL **2.1**
Failed to initialize OpenGL loader!
[WARN] Could not init using OpenGL with normal GLSL version, falling back to GLSL 1.2
Failed to initialize OpenGL loader!
[ERROR] Failed to initialize OpenGL with GLSL 1.2

OpenGL 2.1 is supported by OpenSource Mesa driver in the OS. SDR++ is after figuring out that OpenGL 3.x is not supported falling back to OpenGL 2.1 -> Perfect. That should work now. But it does not. This does not make sense when the supported OpenGL version is exactly the one SDR++ is falling back to. I highlighted the version numbers to make it more clear.

AlexandreRouma, 2024-02-07:
PC too old, not all OpenGL 2.1 are made equal and yours doesn't support GLSL 1.2. (you haven't filled out the template even though it clearly says it's not optional so I'm not gonna be able to help more than that).

closed 2024-02-07 by AlexandreRouma

fizr5, 2024-02-08:
Lets see if the developer now reopen this bug report and fix the problem after years of letting users alone and closing their bug reports. More information here: https://github.com/AlexandreRouma/SDRPlusPlus/issues/1325