AlexandreRouma/SDRPlusPlus #1323: SDR++ Server UDP/TCP

KubaPro010, 2024-02-07:
could a switch to like switch between TCP and UDP (for iq data) be added? so something like the client connects with tcp and its mostly done normally, but if the client would send a packet to switch the protocol, the sdrpp server could open a new server on like one port number higher than the normal port and so it would like send iq data with udp instead of tcp (becasue tcp can cause delay)

AlexandreRouma, 2024-02-07:
Extensive testing was done, UDP is almost always worse over the network for this use than TCP since a packet arriving late is less problematic for the DSP than a packet not arriving at all.

closed 2024-02-07 by AlexandreRouma

KubaPro010, 2024-10-06:
Sure, however doesn't TCP delay add up over time? and also, there's no way that someone is gonna use the server over the internet, in my time i've never seen a single packet drop on my lan, also what was the testing?

AlexandreRouma, 2024-10-06:
It does not. We have run testing with using SDRs in the US from Europe and vice versa with no issues whatsoever.