AlexandreRouma/SDRPlusPlus #1331: Rigctl Client as secondary receiver

magellan-13016, 2024-02-18:
Hello,

I have a SDR TX/RX switch which allow me to use my SDR device as a second receiver and not a Pan adapter. Actually, there is only one way to control the frequency of my Yaesu FT-847 transceiver : acting as a panadapter with the need to set up the IF frequency. 

I just wish you add a way to put the frequency we select in SDR++ to the transceiver via Hamlib so I can ear stations with my SDR device and answer to it with my Yaesu FT-847 transceiver without having to enter every time, the frequency I  choose before.

I've already made this request many times by e-mail and I'm insisting again here because your software is the only one that lets me work comfortably under Linux and I really need this functionality for an educational project with my pupils that will take place next June. 

Best regards, David.

bturbeville, 2024-04-02:
I have a plugin that I have developed and have been testing that implements bi-directional VFO tracking using Hamlib on an IC7610.  I developed it to use WFView with SDR++.  It seems to be working without issue if you would like to build it I can provide the source.  I haven't put in a pull request for this and another enhancement I have made yet.

magellan-13016, 2024-04-07:
Of course I'm interested ! All my request are ignored and if your plugin works for Icom, I hope I will be able to adapt it for my Yaesu FT-847. After that, I will put this plugin on my website for all Yaesu FT-847 Owners... If I succeed. Many thanks for your help.

gbeyer3, 2025-03-21:
Hi @f4bpp, did you have success with that collaboration with @bturbeville ?   I want to do siimilarly with my FT-950.  Thanks!