AlexandreRouma/SDRPlusPlus #1340: No signal only white noise with RTLsdr v4

Schrille, 2024-02-25:
I have downloadet sdrpp but only noise.
No signals received on FM broad band.direct sampling is of, gain is high, mod is WFM

In sdr# every thing works fine.
I guess sdrpp does not work with my dongle

[image]
any suggestion's?🤷🏼‍♀️

kmp56, 2024-02-26:
https://www.rtl-sdr.com/V4/

" Download the Releases.zip file from our [RTL-SDR Blog Driver GitHub releases page](https://github.com/rtlsdrblog/rtl-sdr-blog/releases). 
   - Copy the rtlsdr.dll file into the folder of the software you are using. Usually this will be replacing an existing rtlsdr.dll file.
   - On some computers you may also need to copy the msvcr100.dll and pthreadVC2.dll file into the folder too. It does not hurt to copy these in anyway, even if you don't need them.

If for some reason the RTL-SDR Blog release is not working for you, please try the latest Osmocom build at https://ftp.osmocom.org/binaries/windows/rtl-sdr/ which is now also compatible. "

Schrille, 2024-02-27:
thanks, copy these .dll files that reallydid the trickVon meinem/meiner Galaxy gesendet

https://www.rtl-sdr.com/V4/

closed 2024-02-28 by AlexandreRouma