AlexandreRouma/SDRPlusPlus #1341: DRM Decoder Plugin

var-IO-reeks, 2024-02-27:
**Feature description**
I think a Digital Radio Mondiale (DRM) decoder plugin would be a nice addition to SDR++, since the current standalone DRM programs all have issues and aren't very user friendly.
Using them as a base for a plugin within the already feature-rich ecosystem of SDR++ would solve a bunch of their problems (limited source options, unstable GUI etc.).

Apart from the now pretty much abandoned DREAM DRM decoder, this projects looks for example like a possible base for such a plugin, since it is also written in C++:

https://github.com/JvanKatwijk/drm-receiver

I would love to try to port that by myself, but because of my extremely limited C++ knowledge, that will be a "difficult" task 😬

Axpelle, 2024-03-01:
Agreed. But the suggestion for a built-in plugin is still worth considering. Third party DRM solutions tend to be flaky at best.

AlexandreRouma, 2024-03-07:
> But the suggestion for a built-in plugin is still worth considering.

I would have closed the issue if it wasn't