AlexandreRouma/SDRPlusPlus #1343: Cannot install from .deb SDR++ Nightly on MXLinux v 23.2 Libretto

gstalnaker, 2024-02-28:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel i7-10700 (16 core)
- RAM: 16 Gb
- GPU: NVIDIA X Driver  525.147.05[/CODE]
- SDR: (Remote or local? If remote, what protocol?) Local RTL-SDR USB dongle

**Software**
- Operating System: MXLinux v23 Libretto (based on Debian bookworm)
- SDR++: Cannot install

**Bug Description**

Attempting to install ANY version of SDR++ on this system results in the package manager reporting the following:

```
This action will also remove the following packages:

libasound2-plugins (1.2.7.1-1)
libjack-jackd2-0 (1.9.21~dfsg-3)
libjack-jackd2-0 (1.9.21~dfsg-3)
wine-stable (9.0.0.0~bookworm-1)
wine-stable-i386 (9.0.0.0~bookworm-1)
winehq-stable (9.0.0.0~bookworm-1)
```
Debian sid and buster install attempts fail with dependency errors.

I confirm that if I continue with the install SDR++ works perfectly. But doing so removes Wine, which I require for other purposes. The RTL-SDR dongle works perfectly with another SDR app gqrx, so this is a software conflict I cannot resolve.

During my unsuccessful build from source attempt (see separate bug report), one of the required libraries is rtaudio-dev. I could not install that library via package manager. Doing so results in the removal warning I report above. I conclude that it is this rtaudio-dev package that causes the issue above.

I know this is not technically a bug report as i can't even get the app installed, but I though you should know that for at least one user the .deb packages cannot be installed on their MXLinus v23 Libretto (based on Debian bookworm) system because of this rtaudio-dev issue. I know you have no control over how package managers are configured that has created this issue for me, but knowing that it has happened may help you if other people encounter this as well.

I tried to install from source, but no joy there either.

**Steps To Reproduce**
1. Any attempt to install results in above described behavior.
2. From source build attempt, I know the issue is the rtaudio-dev package - if I attempt to install that (as I did to set up source build dependencies) the package manager reports the same uninstall requirement for wine, etc.

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**

gstalnaker, 2024-02-29:
UPDATE - is was indeed the rtaudio-dev dependency. I used Mr. Google to see if there is a way to disable dependencies and I found a method to edit the .deb contents file. Since I have rtaudio-dev installed as part of the build from source I figured I could remove it from the .deb dependencies. I now have a working SDR++, for which I'm grateful (it is an excellent application you've written). This does not, however, alleviate the issues with the dependencies existing in the .deb package downloaded from the SDR++ website for MXLinux v23 - attempting to install it results in the warning I documented above.

AlexandreRouma, 2024-02-29:
How exactly is this an SDR++ issue? If the maintainers of the RtAudio package on your distro don't know how to package it without nuking half of audio related packages off your system, there's nothing I can do about it. SDR++ uses RtAudio for output, that's it. i'm not gonna change what audio abstraction library I use because of the skill issue from some distro maintainers. Just on their bug report platform and say that installing `rtaudio-dev` removes half of the system's audio components.

closed 2024-02-29 by AlexandreRouma

gstalnaker, 2024-02-29:
Alexandre - I specifically said this was not an SDRPP issue. I submitted solely FYI in case you had reports from other users of this behavior. I did not think (DOH!) to report this to the upstream package maintainers as you suggest. I will do so.