AlexandreRouma/SDRPlusPlus #1351: Scanner Module - Save Preferences.

anathn, 2024-03-07:
Just a simple update to the scanner module so it saves user prefs to config to prevent the user from having to reconfigure it every program launch.  

Since its clear PR's are not welcome, just thought I would offer this up. Feel free to ignore/close this.

https://github.com/anathn/SDRPlusPlus/tree/feature/scanner-pref-save

AlexandreRouma, 2024-03-07:
There's a good reason for it not saving configs. As the readme states, it's still in beta and a lot is expected to change, I don't want to have to implement config conversions once the final module is written.

anathn, 2024-03-07:
That makes sense. I'll close this - it just felt rude for me to not offer the work in case you might want it.

closed 2024-03-07 by anathn

AlexandreRouma, 2024-03-07:
Don't worry, It's not rude, I actually prefer implementing things myself.